yozik04/nibe #1: TCP Modbus s1255

krito, 2022-05-17:
First of all, thank you for your contribution!
I used your example code in readme.md
I tested the TCP modbus with my Nibe S1255 and I had to change read_coil to read_input_registers in the modbus.py file. 

S1255 only have registers. And I had to change in the f1255 json-file to match one address. And it worked.

I did not test writing to a holding register.

yozik04, 2022-05-18:
I have not tested it because I have F1255 which does not have tcp modbus. Please make a PR.
I also do not know the difference between coils and registers in modbus so I am sure your fix is right.

yozik04, 2022-05-29:
I have released 0.2.0 with the proposed changes.

closed 2022-05-29 by yozik04