yozik04/nibe #32: Install in HassOS SMOS40+S2125

cybermaus, 2022-10-07:
The new HA version made me aware of this project.
At this moment, I am directly using the modbus integration to get some values, but this one seems very promising.
But alas, I cannot use the official integration, because I have a newer S series, and tcp/modbus.

Which seems to be listed here as experimental. Great.

I see some install instructions for Docker and PyPi, but not for HA OS.
Can anyone give me some pointers on how to install and try this?

yozik04, 2022-10-07:
@elupus I have some CSV data for you in English =)

nkrepo, 2022-10-16:
Hi all,
I have the SMOS40 controller connected  with F21210 heat-pump.
I am attaching the the Modbus registers dumped from the SMOS40 controller with firmware Version 2.14.8.

As I can understand through the Modbus of SMOS40 controller the registers from F21210 heat-pump can be accessed also.

[smos40_modbus_addresses_all_20221016-06765422020001-1_firmware_2.14.8.csv](https://github.com/yozik04/nibe/files/9794245/smos40_modbus_addresses_all_20221016-06765422020001-1_firmware_2.14.8.csv)

cybermaus, 2022-10-16:
If you are using Home Assistant, pending the completion of this project, (and assuming you know how to modify the yaml configuration) you can use this example to get out some values into your HA.
([which I copied and modified from here](https://community.home-assistant.io/t/modbus-configuration-for-nibe-s-series-heatpumps/400422/4))

```
modbus:
   - name: "Nibe2125"
     type: tcp
#    host: 192.168.x.x # over WiFi
     host: 192.168.x.x # over Eth
     port: 502
     
     sensors:
#Priority	MODBUS_INPUT_REGISTER	1028	1		4	0	0	0
        - name: "WP Operating Prioritisation"
          unique_id: "wp_operating_prioritisation"
          data_type: int16  # int8
          # 10: Off, 20: Hot Water, 30: Heat, 40: Pool, 50: Cooling
          address: 1028
          input_type: input  # R/W
          scale: 1
          precision: 0
          state_class: measurement
          slave: 1
        - name: "WP Outdoortemperatur (BT1)"
          unique_id: "wp_outdoor_temp_bt1" 
          device_class: temperature
          unit_of_measurement: "°C"
          data_type: int16
          state_class: measurement
          address: 1
          count: 1
          slave: 1
          precision: 1
          input_type: input
          scale: 0.1
          scan_interval: 60
        - name: "WP AVG Outdoortemperatur (BT1)"
          unique_id: "wp_avg_outdoor_temp_bt1" 
          device_class: temperature
          unit_of_measurement: "°C"
          data_type: int16
          state_class: measurement
          address: 37
          count: 1
          slave: 1
          precision: 1
          input_type: input
          scale: 0.1
          scan_interval: 60
#Roomsensor 1-1	MODBUS_INPUT_REGISTER	26	10	°C	2	0	0	0
        - name: "WP Room temperature 1 (BT50)"
          unique_id: "wp_room_temp_bt50"
          device_class: temperature
          unit_of_measurement: "°C"
          data_type: int16
          state_class: measurement
          address: 26
          count: 1
          slave: 1
          precision: 1
          input_type: input
          scale: 0.1
          scan_interval: 60
#Room average temp. clim. system 1 (BT50)	MODBUS_INPUT_REGISTER	116	10	°C	2	0	0	0
#RMU is 2*AVG-BT1 (or rather, AVG=(BTU+RMU)/2
        - name: "WP Room temperature 2 (RMU1)"
          unique_id: "wp_room_temp_rmu"
          device_class: temperature
          unit_of_measurement: "°C"
          data_type: int16
[... 191 lines cut ...]
          scan_interval: 60

#Heating medium pump speed (GP1)	MODBUS_INPUT_REGISTER	1636	1	%	4	0	0	0
        - name: "WP Speed loadpump (GP12.1)"
          unique_id: "wp_speed_loadpump1"
          device_class: speed
          unit_of_measurement: "%"
          data_type: int16
          state_class: measurement
          address: 1636
          count: 1
          slave: 1
          precision: 1
          input_type: input
          scale: 1
          scan_interval: 60

        - name: "WP AUX11 Active Heating"
          unique_id: "wp_aux11"
          data_type: int16
          address: 224
          input_type: input
          state_class: measurement
          scan_interval: 60
```

yozik04, 2022-10-18:
@nkrepo Testing is missing =)

nkrepo, 2022-10-18:
> @nkrepo Testing is missing =)

I am happy to assist then ;).

yozik04, 2022-10-18:
@nkrepo you can try to use nibe-mqtt hassio addon then. It might work =)
@elupus will start working on adding modbus to HA integration.

cybermaus, 2022-10-19:
@elupus So I exported the registers again. First, simply because I activated some options, so lets see.
Next I did the firmware, and exported yet again.

So, it is true, what you connect modifies the list of registers
Since previous export, I installed BT7, also hooked something up to BE6 and BE8, as well as activated SG-Ready and Energy-pulse counters, and got a bunch of new values in the same firmware. Note that BT7 was also added to the default list, but these were added to the all list:
(The Operating time and Downtime actually surprise me. Surely they should have been there before?)

```
Hot water top (BT7)	MODBUS_INPUT_REGISTER	8	10	°C	2	0	0	0
Pulse energy meter (BE6)	MODBUS_INPUT_REGISTER	396	100	kWh	6	0	2147483647	0
Pulse energy meter (BE8)	MODBUS_INPUT_REGISTER	398	100	kWh	6	0	2147483647	0

Operating time	MODBUS_HOLDING_REGISTER	3102	1	min	1	1	60	60
Downtime	MODBUS_HOLDING_REGISTER	69	1	min	1	0	60	0
Cooling (SG Ready)	MODBUS_HOLDING_REGISTER	761	1		4	0	1	1
Hot water (SG Ready)	MODBUS_HOLDING_REGISTER	762	1		4	0	1	1
Operating mode circulation pump hot water heat pump 8	MODBUS_HOLDING_REGISTER	792	1		4	0	1	1
Operating mode circulation pump hot water heat pump 7	MODBUS_HOLDING_REGISTER	793	1		4	0	1	1
Operating mode circulation pump hot water heat pump 6	MODBUS_HOLDING_REGISTER	794	1		4	0	1	1
Operating mode circulation pump hot water heat pump 5	MODBUS_HOLDING_REGISTER	795	1		4	0	1	1
Operating mode circulation pump hot water heat pump 4	MODBUS_HOLDING_REGISTER	796	1		4	0	1	1
Operating mode circulation pump hot water heat pump 3	MODBUS_HOLDING_REGISTER	797	1		4	0	1	1
Operating mode circulation pump hot water heat pump 2	MODBUS_HOLDING_REGISTER	798	1		4	0	1	1
Operating mode circulation pump hot water heat pump 1	MODBUS_HOLDING_REGISTER	799	1		4	0	1	1
Mode, energy meter (BE6)	MODBUS_HOLDING_REGISTER	4015	1		4	0	255	255
External energy meter 1 pulse per kWh	MODBUS_HOLDING_REGISTER	4017	1		5	1	10000	500
External energy meter 2 pulse per kWh	MODBUS_HOLDING_REGISTER	4018	1		5	1	10000	500
External energy meter 3 pulse per kWh	MODBUS_HOLDING_REGISTER	4119	1		5	1	10000	500
Mode, energy meter (BE11)	MODBUS_HOLDING_REGISTER	4127	1		4	0	255	255
Start time HWC period 1	MODBUS_HOLDING_REGISTER	4024	1		-	-	-	-
Start time HWC period 2	MODBUS_HOLDING_REGISTER	4026	1		-	-	-	-
Start time HWC period 3	MODBUS_HOLDING_REGISTER	4028	1		-	-	-	-
Stop time HWC period 1	MODBUS_HOLDING_REGISTER	4030	1		-	-	-	-
Stop time HWC period 2	MODBUS_HOLDING_REGISTER	4032	1		-	-	-	-
Stop time HWC period 3	MODBUS_HOLDING_REGISTER	4034	1		-	-	-	-
id:7855	MODBUS_HOLDING_REGISTER	4036	1		4	0	1	0
id:7856	MODBUS_HOLDING_REGISTER	4038	1		4	0	1	0
id:7857	MODBUS_HOLDING_REGISTER	4039	1		4	0	1	0
Activate SG Ready via API	MODBUS_HOLDING_REGISTER	3032	1		4	0	1	0
```
Next I updated the firmware. One value changed, and I got a buch of new ones.
Old firmware: 2.13.5 (I think, from memory)
New firmware: 2.14.8

The changed one (the unit kW was removed, not sure this is supposed to be an improvement)
```
Power at DOT, manual value	MODBUS_HOLDING_REGISTER	4200	1		4	0	1	0
```
And the added lines to the new firmware
```
id:4591	MODBUS_HOLDING_REGISTER	4052	1		1	5	80	30
id:13796	MODBUS_INPUT_REGISTER	2243	1		4	0	0	0
id:13797	MODBUS_INPUT_REGISTER	2244	1		4	0	0	0
id:13798	MODBUS_INPUT_REGISTER	2245	1		4	0	0	0
id:13799	MODBUS_INPUT_REGISTER	2246	1		4	0	0	0
id:13800	MODBUS_INPUT_REGISTER	2247	1		4	0	0	0
id:13801	MODBUS_INPUT_REGISTER	2248	1		4	0	0	0
id:13802	MODBUS_INPUT_REGISTER	2249	1		4	0	0	0
id:13803	MODBUS_INPUT_REGISTER	2250	1		4	0	0	0
Disable emergency restart GP1	MODBUS_HOLDING_REGISTER	4204	1		4	0	1	0
Disable emergency restart GP1	MODBUS_HOLDING_REGISTER	5025	1		4	0	1	0
id:24483	MODBUS_HOLDING_REGISTER	5008	1		4	0	1	0
id:24484	MODBUS_HOLDING_REGISTER	5009	10	°C	2	50	800	450
id:24485	MODBUS_HOLDING_REGISTER	5010	10	°C	2	50	800	450
id:24486	MODBUS_HOLDING_REGISTER	5011	10	°C	2	50	800	450
id:24487	MODBUS_HOLDING_REGISTER	5012	10	°C	2	50	800	450
id:24488	MODBUS_HOLDING_REGISTER	5013	10	°C	2	50	800	450
id:24489	MODBUS_HOLDING_REGISTER	5014	10	°C	2	50	800	450
id:24490	MODBUS_HOLDING_REGISTER	5015	10	°C	2	50	800	450
id:24491	MODBUS_HOLDING_REGISTER	5016	10	°C	2	50	800	450
id:24492	MODBUS_HOLDING_REGISTER	5017	10	°C	2	-50	300	250
id:24493	MODBUS_HOLDING_REGISTER	5018	10	°C	2	-50	300	250
id:24494	MODBUS_HOLDING_REGISTER	5019	10	°C	2	-50	300	250
id:24495	MODBUS_HOLDING_REGISTER	5020	10	°C	2	-50	300	250
id:24496	MODBUS_HOLDING_REGISTER	5021	10	°C	2	-50	300	250
id:24497	MODBUS_HOLDING_REGISTER	5022	10	°C	2	-50	300	250
id:24498	MODBUS_HOLDING_REGISTER	5023	10	°C	2	-50	300	250
id:24499	MODBUS_HOLDING_REGISTER	5024	10	°C	2	-50	300	250
```

elupus, 2022-10-19:
Can you upload the full file here. I have another pull up for some updates.

cybermaus, 2022-10-19:
[modbus_addresses_all_20221019-06765422110029-2.csv](https://github.com/yozik04/nibe/files/9817411/modbus_addresses_all_20221019-06765422110029-2.csv)

cybermaus, 2022-10-19:
The above one is after I had disabled the Pulse counters, so those are not in there anymore.
This is the file still with older firmware, but with the pulse counters still active:
[modbus_addresses_all_20221019-06765422110029-1.csv](https://github.com/yozik04/nibe/files/9817419/modbus_addresses_all_20221019-06765422110029-1.csv)

nkrepo, 2022-10-19:
I performed a diff between the modbus_addresses_all_20221016-06765422020001-1_firmware_2.14.8.csv  and  odbus_addresses_all_20221019-06765422110029-2.csv files. My SMOS40 has firmware version 2.14.8.

Here are the differences:
[image]

I have the pulse meter enabled,  i am not really sure what the deal is with the "Hot water additional heat used in heating mode" and I don't have room temperature sensor.

@elupus I see you have added the F2120 registers in smos40.csv, smo40s0.json so I suppose we are good to go.

[nibe_SMOS40_F2120_vs_S2125.txt](https://github.com/yozik04/nibe/files/9818766/nibe_SMOS40_F2120_vs_S2125.txt)

elupus, 2022-11-11:
I'll close this now as it should be supported. Open new issues if something remain.

closed 2022-11-11 by elupus

cybermaus, 2022-11-16:
Using this closed issue for some side-topic: @nkrepo as the only person I know who also has a SMO S40, I have a question for you:

I am noticing my SMO S40 "hangs" fairly often. Once I had to unplug it, after it did not give any data for 4 hours, and also the screen was jammed. But also, every so often (let's say on average once a week, but very randomly spread) I see evidence it hung and restarted.

The evidence is that it is that this plugin missing data for a few minutes. But also some direct modbus reads have a start temp of -3267.8C. And the BT1 rolling average temperature resets to current BT1 value.
And this morning I happened to catch it in red-handed, as the screen was just rebooting as I walked past it.

I suspect it hangs, and after a few minutes some watchdog timer resets it. 

So @nkrepo: Are you seeing the same?

[image]

[image]

[image]

cybermaus, 2022-11-22:
@nkrpo Just the build-in modbus integration. I sent you the entire source earlier.

It hung and restarted again yesterday. I wonder (in fact I hope) it is the barrage of modbus requests that is causing this. After all, I am reading the modbus double, once with the native modbus, once with this advanced NIBE integration. But is it caused by it, and if so, which one?

I guess if you do not have any integration active, it is very difficult to detect it hangs at all. Unless if you happen to catch it in the act, like I did twice now in the last 2 months.

But for now, as I test, since yesterday I have completely disabled the modbus integration, but left the NIBE integration active. Lets see for 2 weeks or so, and then I will swap them.

nkrepo, 2022-12-08:
> @nkrpo Just the build-in modbus integration. I sent you the entire source earlier.
>
> It hung and restarted again yesterday. I wonder (in fact I hope) it is the barrage of modbus requests that is [...]
@cybermaus 
I updated latest firmware version 2.15.9 and added some modbus registers through the updated NIBE integration, I will monitor for any SMOS40 crashes from now one. It is possible that some short of memory leak exists in NIBE modbus process and frequent polling of the modbus registers may lead to crashes.

cybermaus, 2022-12-08:
Thanks. Mind you, I have no reason to think it is the modbus. I was just wondering especially since I was reading the modbus twice, using two mechanisms.

At this moment, I stopped one of the two mechanisms, and it does not matter at all, Still a crash every 4~6 days.
I want to stop all modbus, as a test. But the problem is, how do I then know it is crashing. I would not know, as I would not get any of the sensor readings to see the crash. 
In fact, I wonder if *everyones* SMO S40 crash every so often, and simply no one is noticing.

I think I need to hook up a sensor to AUX10 and AUX11, one that is reasonably speaking always on, and then using that monitor for a restart (as the AUX10 relay and AUX11 relay would both briefly turn of at the same time during restart.)

nkrepo, 2022-12-27:
@cybermaus I have not noticed any crashes the last three weeks
For example here is the log from the supply line
[image]

cybermaus, 2022-12-27:
Thanks. And while it is quite possible it does not crash for you, you would not notice it on the BT12 supply line. Not with this yozik/elupus method. They filter out the invalid -3200C. And you would likely not miss 1 or 2 missing values.

Using there interface, I only see the crash using the BT1 average outdoor temp. What happens when it crashes (or even if you restart manually, is the average temperature jumps to the current outdoor temperature. So whenever that happens (and I did not restart it myself) I know it did a watchdog restart.

Mind you, I find your BT12 rather interesting. It looks soo much different. 
Both change behaviour on Dec 19 though, That is when the cold weather stopped for me, I assume for you too.
Where (what country) are you located? 

[image]

nkrepo, 2022-12-28:
I am located at Greece about 500m altitude, bellow you can see my BT1 and average BT1, i think my average BT1 does not jump at any point so I think it does not crash.
[image]

The shape of my BT12 is formed from the thermostats which turn on-off the heating , I upgraded from a boiler heater to heatpump/buffer system. The usage of thermostats that turn on-off the water supply to the radiators is less than ideal I think for a heatpump system. 
I am investigating if there are any gains to switch to a continuous operation system by regulating the water temperature at the buffer's output and install NIBE's temperature sensors so the radiators are supplied with just enough warm water to keep the temperature stable.

 How your systems controls the temperature for each zone?

cybermaus, 2022-12-30:
I agree, no evidence of reset. Hmm, maybe I'll do a factory reset.
Do you have a buffer vat, or is the S2125 feeding the radiators directly? (sorry if you already told)

I only have one zone, with a NIBE RMU S40 controlling it from measuring the central living room. All downstairs radiators and underfloor are fully open, all upstairs one fully closed. I did check the radiators and underfloor restrictors though, to ensure they all drop the same amount of temperature.

Besically the other rooms stay warm enough. Even the upstairs, as heat flows up anyway.

For the rest, I kept the SMO S40 settings pretty much default, and it seems to want to heat up slowly and gradually, hence my relatively low medium temperature. Mind you, when the whether dropped to -5C for a week, it was coping but it did had to defrost quite often.