yozik04/nibe #38: HA state_class & device_class

yozik04, 2022-10-10:
@elupus, do we want to have state_class and device_class attributes for every coil in json files? You will need them in HomeAssistant integration.

yozik04, 2022-10-10:
No I do not offer to add these manually. We can try to detect these from `unit_of_measurement`.

yozik04, 2022-10-10:
This is how I did it in https://github.com/yozik04/nibe-mqtt/commit/aa83e4760921e2603c340a8e312824ef358a2197

elupus, 2022-10-10:
We already do that in the HA integration. Those classes are internal to HA so they should not really be in lib.

tanelvakker, 2022-10-11:
There are a lot of registers that don't have a unit defined. One example is compressor starts. Currently it ends up without a class and is considered a text field by HA.
  "43635": {
    "title": "EB108-EP14 Compressor starts",
    "size": "u32",
    "factor": 1,
    "min": 0.0,
    "max": 9999999.0,
    "default": 0.0,
    "name": "eb108-ep14-compressor-starts-43635"
  },
[image]

elupus, 2022-10-11:
We can mark them here with a custom unit, and translate that in HA.

tanelvakker, 2022-10-11:
Would adding state_class "measurement" for any integers be a solution that takes care of most instances?

elupus, 2022-10-11:
Yes, but we also need to to get total/increasing in place. I do think it would be good if we added some unit in the lib.

tanelvakker, 2022-10-11:
If you can define what the units are for counter type measures, I can start manually adding these to jsons. Would compressor starts have a measurement "times" or maybe "#"?

elupus, 2022-10-11:
Or count. However i do think we should have a way of specifying extensions to the raw csv file, that can be applied to multiple pump types. @yozik04 thoughts?

yozik04, 2022-10-11:
raw csv files are unmodified files exported from a heatpump or Modbus Manager. I do not want to touch these. We can edit json files if needed. I will need to store manual patches to json files some day, somehow.

@tanelvakker compressor starts does not need a unit. It requires state_class=total, I think. https://developers.home-assistant.io/docs/core/entity/sensor/#how-to-choose-state_class-and-last_reset

elupus, 2022-10-12:
@yozik04 yes, but we currently base those data on unit availability. We could as mentioned add state class and device class to the json file. But it does feel very home assistant specific, so i'm a bit hesitant. I did this to move changes to a separate extension file: https://github.com/yozik04/nibe/pull/40 but maybe it should be a per-device extension file.

elupus, 2024-02-06:
Shown up again: https://github.com/home-assistant/core/issues/104074 we need to figure out some way to handle this.