yozik04/nibe #41: Is the SMO 20 really supported?

storeman, 2022-10-13:
So I'm trying to find if I can locally connect to my SMO 20 / F2120. It is mentioned on some places:
- the yozik04/nibe repo
- Home Assistant integration, which seems to use your list

If I dig in to the documentation, I go to the NibeGW docs. Here the SMO 20 is not listed anywhere.

I would really love to use the SMO 20 with an ESP/Arduino, but I think it is impossible. I can't find anything about any modbus or RS485 in the installer manual from the SMO 20 / F2120.

elupus, 2022-10-14:
Can you enable the modbus accessory if you enter the setup menu on the pump? If you can it should be possible to use. Otherwise it's likely not.

storeman, 2022-10-14:
The heat pump has no settings for MODBUS in the technical documentation and no connectors for this. Also, the control unit SMO 20 doesn't have any connectors or menu options. Should I create a PR to remove this from the code?

elupus, 2022-10-14:
Can you also check the actual menus of the pump?

Ps. I suspect one could hook up to same communication line that the smo connects to the actual pump.

storeman, 2022-10-14:
What do you mean by the "actual menus of the pump"? In my previous two posts I mentioned that I've looked through the menu of the SMO 20.

I've digged through the menus a couple of times and things about MODBUS or serial connection is not mentioned anywhere. If you have a suggestion where I could look, I'm happy to do so.

elupus, 2022-10-14:
You have opened up the service menu by holding down the back button for a few seconds (at least that is the common method) and got to the list of enabled accessories?

storeman, 2022-10-14:
Yes, that's where the good stuf is at :).

yozik04, 2022-10-14:
I do not agree with you. Why would nibe include smo20 in Modbus Manager if modbus would not be supported?

yozik04, 2022-10-14:
[image]

storeman, 2022-10-14:
That is curious. I very much hope to connect to it using Modbus, I see the SMO20 in the application too, I just downloaded it. But I really don't have a clue what and how to connect to it physically.

If I check the MODBUS40 accessory, the SMO20 is not mentioned. Just connecting something between the SMO20 and the F2120 doesn't make much sense, then they both would support it and would be choosable on the modbus manager. Also, why would the SMO40 need an extra accessory to support it.  

I already spend a couple of hours researching. I just want to prevent people getting theirs hopes up or spending hours searching for something which is not possible.

elupus, 2022-10-14:
The connection btween pump and smo is a rs485 bus. It can be daisy chained with multiple devices. So fully possible to listen to data there. But i we cant enable the accessory, it wont send data needed to request parameters.

I agree, strange its in modbus manager. But could be a future proofing. The parameter sets are pretty much identical between pumps.

yozik04, 2022-10-14:
You can send a message directly to Nibe and ask them. They are quite responsive.

storeman, 2025-10-06:
I haven't researched this any further. I've installed an SMO S40.

closed 2025-10-06 by storeman