yozik04/nibe #45: Reject absurd values on SMO S40

cybermaus, 2022-10-18:
I open this in a new issue, not sure if I am doing right on that. Maybe you want it direct to one of the other S40 threads.
And in fact, I am not even sure if it is an issue in this software, maybe you already do this value sanitation.

But I read the MODBUS directly in HA and I noticed, every time I turn off/on the device, some temperatures' first reported value is -3276.8 Celcius. Scientist would love to get their hands on this device :)

It rather messes up the graphs though. So far, I have removed the values with SQL (see below) but maybe you can pre-empt this and reject any temperatures below 0 Kelvin or something.

[image]

`SELECT * FROM "states" where entity_id like "%BT%" and entity_id like "%temp%" and cast(state as float)<-1000`
```
state_id | domain | entity_id | state
40001087 | NULL | sensor.wp_supply_line_temperature_bt12 | -3276.8 
40001088 | NULL | sensor.wp_return_line_temperature_bt3 | -3276.8
40001151 | NULL | sensor.wp_avg_outdoortemperatur_bt1 | -3276.8
```
`DELETE FROM "states" where entity_id like "%BT%" and entity_id like "%temp%" and cast(state as float)<-1000`
`DELETE FROM "statistics_short_term" 
where exists (
select * from statistics_meta B
where metadata_id=b.id and 
b.statistic_id like "%BT%" and b.statistic_id like "%temp%" 
and cast(mean as float)<-100 )`

cybermaus, 2022-10-18:
Edit: I suddenly recognise the value as int16  x"FFFF" (divided by 10)
Though probably that does not matter much.

elupus, 2022-10-18:
Its already handled :)

closed 2022-10-18 by elupus

elupus, 2022-10-18:
@cybermaus would you be able to test out: https://github.com/home-assistant/core/pull/80557?

cybermaus, 2022-10-18:
Probably. I'd have to figure out how to load a pre-release core onto my HA-OS.   Never done that yet
With some google I will probably be able to figure it out, and don't put in too much effort, but if you have a link or something handy, that might help.

Right at this moment though, I am busy soldering a kWh pulse reader (EB1 extention) onto the NIBE, so it may be later this evening or tomorrow only.

cybermaus, 2022-10-18:
The S40 has a configurable energy pulse output. Called BE1 in the installer manual, best study that for your F1255. If I can read those with ESPHome, we can move it to Home Assistant. (Note: not successful so far)

Mind you, MODBUS also has kWh output, alas not working on my S40. But maybe it is working for your F1255, in which case you get it for free.

cybermaus, 2022-10-18:
Its called (BE6, BE7) on clamps (X22, X23) on your F1255. If/when I get it working, I'll let you know. Note however, not sure where you guys get the time for all of this work that I see happening, but I cannot spend too much time. It may be a while.

elupus, 2022-10-18:
@cybermaus i think you have missunderstood those. They are for connecting an external meter, that has a pulse output to the pump, so the pump knows energy usage. The pump does not know that itself. Ie, it's an input, not an output.

cybermaus, 2022-10-18:
@elepus .... Are you sure? I did realise you can also connect the current measurement coils to the same clamps, and it clearly states in my manual you either put energy coils, or pulse meters, but not both.
Now you mention it, the whole section on pulse energy meters is indeed ambiguous. It does not mention input our output. 

I understand the use of the current coils as input. It is for the backup heater, so it can decide which stages to activate without blowing a fuse. But what possible use would a Energy/kWh meter be as input to the heat pump? 

Though I now worry you may be right, and it surely would explain why I am unable to see any pulses. But again, what possible use is a Energy input (not even power or current, but energy) for this device ?!?

elupus, 2022-10-18:
Its required by some countries for the heater to be able to report the total consumption of the pump,.but its not required in some countries. So building it in is wastefull.

cybermaus, 2022-10-18:
Well, given the shared input on the EMK (which are simple pulse flow meters) and the weird +5v / pulse connection schematic (rather then pulse/Gnd) I have to concede, these are inputs, not outputs. (oh, and not to mention I am not measuring anything on my scope)

Well, lucky it was only a partial reason I hooked up an ESPHome. The other was, I am putting SG Ready on the AUX and also I am dynamically controlled PWM speed on the GP10 pump based on delta temp. That is at least working. (now to figure what is the best delta temp. (note the GP10 is not directly connected, there is a UKV buffer inbetween)

yozik04, 2022-10-19:
@cybermaus Isn't there an option in settings menu for your pump to automatically control supply and brine pumps speed? At least this is what I use. Or is this some additional pump?

cybermaus, 2022-10-19:
Yes, the GP12.1 load pump is automatically controlled. It flows from HP to Buffer, and I am not touching that, as I assume the NIBE algorithms are skilled. (BTW: mine is a Air/Water HP, no brine)

The external GP10 pump from the Buffer to the radiators and floor heating (the external circuit in NIBE speak) however, is a simple on/off control. Even though there is a PWM drive on the S40 board, it is not used. GP10 is always at full speed, not only making noise, but also drawing ~60W, even when there is no heating, 24/7.

I made a ESP board to control it, currently it senses delta temperature, and tries to maintain a certain delta temperature by modulating speed. There is no difference in delta T all the way down to 50% (I think the pump it is just fighting and not pumping above that). Also, when no heat is supplied, speed sinks to 5% with only 10W of use and no noise.

It only has been active for a day, and honestly, it is not cold enough yet in Netherlands to call it a success, but so far I am quite pleased with the results, merely based on the lack of standby energy en standby noise when there is no heating.

However, any well documented references on what should be flow rates and delta temperatures you may have I will read and consume.

cybermaus, 2022-10-19:
> @cybermaus would you be able to test out: [home-assistant/core#80557](https://github.com/home-assistant/core/pull/80557)?

@elepus I may need some help

I [found a post](https://community.home-assistant.io/t/how-to-install-forked-core-to-test-integration/309449/2), claiming you could just drop the files directly from a forked `homeassistant/components` into `~/config/custom_components` and then it would pick up the modified component.

So I downloaded the zipped forked commit, dropped `components/nibe_heatpump` into  `~/config/custom_components/nibe_heatpump`, and restarted HA

However, if I now add NIBE integration, it still has the old list of pumps, and also still demands the gateway ports, not the TCP port.

elupus, 2022-10-19:
Just at the end after the definition of iot_class. I added a leading comma in my text since this is a needed separator, but it could be added trailing the iot_class definition instead as the other separators.

elupus, 2022-10-19:
Ive some pending docs update here: https://deploy-preview-24619--home-assistant-docs.netlify.app/integrations/nibe_heatpump/#tcpip

elupus, 2022-10-19:
Make sure you copied the hidden language files (dot prefixed). Also refresh your browser.

cybermaus, 2022-10-19:
Getting an instant Failed (no wait or timeout)
Tried with both 0 and 1 for what I suspect is the slave indivator
With the failed screen, the values reset, so I added before and after screens.
[image]
[image]

elupus, 2022-10-19:
Lets continue in separate issue: https://github.com/yozik04/nibe/issues/50