yozik04/nibe #50: Initial support for modbus is failing

elupus, 2022-10-19:
Initial support for modbus is failing

elupus, 2022-10-19:
@cybermaus can you attach home assistant logs, so we can see the error.

elupus, 2022-10-19:
That was a bit unexpected. Did any of you use standard modbus integration from ha against these pumps? Mainly to get some clues.

cybermaus, 2022-10-19:
Yes. I did stop it, (directed it to another IP) and restarted HA, and then tried again. Same result.

cybermaus, 2022-10-19:
@dkmh where do you find those. All I get is below, not very interesting.

```
Full logs
2022-10-19 14:30:03.690 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration nibe_heatpump which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-10-19 14:30:03.696 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solaredge_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-10-19 14:30:03.699 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-10-19 14:30:29.132 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.total_consumption (kW) can not be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

cybermaus, 2022-10-19:
Never mind, log had wrapped, had to redo the error:
(same as already found)
```
2022-10-19 15:12:23.079 ERROR (MainThread) [custom_components.nibe_heatpump] Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/nibe_heatpump/config_flow.py", line 188, in async_step_modbus
    title, data = await validate_modbus_input(self.hass, user_input)
  File "/config/custom_components/nibe_heatpump/config_flow.py", line 144, in validate_modbus_input
    coil = await connection.read_coil(coil)
  File "/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py", line 69, in read_coil
    result = await self._client.read_holding_registers(
  File "/usr/local/lib/python3.10/site-packages/async_modbus/core.py", line 181, in read_holding_registers
    return await self._send_message(request)
  File "/usr/local/lib/python3.10/site-packages/async_modbus/core.py", line 142, in _send_message
    return await self.protocol._async_send_message(
  File "/usr/local/lib/python3.10/site-packages/async_modbus/core.py", line 43, in send_message_tcp
    tcp.raise_for_exception_adu(response_error_adu)
  File "/usr/local/lib/python3.10/site-packages/umodbus/client/tcp.py", line 247, in raise_for_exception_adu
    pdu_to_function_code_or_raise_error(resp_pdu)
  File "/usr/local/lib/python3.10/site-packages/umodbus/functions.py", line 118, in pdu_to_function_code_or_raise_error
    raise error_code_to_exception_map[error_code]
umodbus.exceptions.IllegalFunctionError: Function code is not an allowable action for the server.
➜  config
```

elupus, 2022-10-19:
@cybermaus can you read the holding register 22 using standard modbus code?

cybermaus, 2022-10-19:
Maybe not, I tried to read it as an input register, and now HA is not restarting. 

In the mean time, I also did a `tcpdump` of the moment when I pressed submit on the plugin, and got this wireshark view. One modbus send, and one modbus reply:

[image]
[image]

[wireshark.zip](https://github.com/yozik04/nibe/files/9821060/wireshark.zip)

cybermaus, 2022-10-19:
Oh wait, it did restart, just took a bit longer for some reason. But I am not reading it correctly appearantly:

[image]
[image]

cybermaus, 2022-10-19:
21 is not in my register list. Howmany bytes? What name?

elupus, 2022-10-19:
22 is likely right. There are some offset to this number that i dont know if its added in the csv export.

cybermaus, 2022-10-19:
tried holding as well, accepted by the startup parser, still unavailable answer.

elupus, 2022-10-19:
Can you check wireshark there too? Hopefully identical.

cybermaus, 2022-10-19:
Yes, it appears all input registers go well, the one holding register fails:

(so, me being a MODBUS noob, what is a holding register?)

[image]
[image]

elupus, 2022-10-19:
Register that can both be read and written. You can test some other of the holding registers and see if you get data (look in csv file).

It could be that its just rejecting this one, with wrong error code.

cybermaus, 2022-10-19:
Well, I am trying `More hot water	MODBUS_HOLDING_REGISTER	697	1		1	0	0	0
` and it gives the same.

To be fair: Out of all the (input) registers I am trying to read in the past few weeks, most of them fail. I come to the acceptance that only a few (lucky the most important) registers work. I assumed the register list is some sort of a "hint" which registers *might* be available, but not a definite list.

Therefore, while I did not think much about it till now, I would have assumed you wanted a full register list so you can try them all, and forget if they do not work. Ignore any read error, and move on to the next.
But now I understand you feel you should be able to read them all. Is this how it works on the non-TCP modbus, they should all exists?

Of course, my other though was, maybe they should all work, just the generic modbus integration of HA is not capable enough? So I am not sure. I was hoping you would read more of them, but to be honest, I was also hoping most would be automatically disabled or hidden in the integration, because I sure do not need to keep history on all of those potential registers. Especially not the ones named "id:" or "-"

cybermaus, 2022-10-19:
But such a readonly setting would not explain the many input registers that I also cannot read.

cybermaus, 2022-10-19:
All hidden by default is still a very big set of data that we are recording. most could/would/maybe even should be disabled, to save the recorder space.

elupus, 2022-10-19:
Not being recorded, i meant disabled.

elupus, 2022-10-19:
Nice!! Then i will fix those offsets and add mapping for that register.

elupus, 2022-10-19:
I've now updated the pull request, so you can regrab it @cybermaus

dkmh, 2022-10-19:
config validation still fails for me:
1:  `coil = heatpump.get_coil_by_name("reset-alarm-40022")`      <--- has to be changed now to 40023
2:  write_coil seems to fail (im not sure what the value code is for)

After changing that i fall into this error (was already a problem in last version, but forgot to mention it):
`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/nibe_heatpump/__init__.py", line 63, in async_setup_entry
        heatpump.word_swap = entry.data[CONF_WORD_SWAP]
KeyError: 'word_swap'
`
But after removing that sensor values seem to work (just checked a few)

dkmh, 2022-10-19:
On changing parameters I get this error:
`Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/nibe_heatpump/__init__.py:224
Integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 12:06:09 AM (3 occurrences)
Last logged: 12:07:07 AM

[1657699968] The value contained in the request data field is not an allowable value for the server.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/umodbus/functions.py", line 1515, in values
    struct.pack(">" + conf.MULTI_BIT_VALUE_FORMAT_CHARACTER, value)
struct.error: required argument is not an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 109, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/nibe_heatpump/number.py", line 68, in async_set_native_value
    await self._async_write_coil(value)
  File "/config/custom_components/nibe_heatpump/__init__.py", line 304, in _async_write_coil
    await self.coordinator.async_write_coil(self._coil, value)
  File "/config/custom_components/nibe_heatpump/__init__.py", line 224, in async_write_coil
    coil = await self.connection.write_coil(coil)
  File "/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py", line 110, in write_coil
    result = await self._client.write_registers(
  File "/usr/local/lib/python3.10/site-packages/async_modbus/core.py", line 233, in write_registers
    request = self.protocol.write_multiple_registers(
  File "/usr/local/lib/python3.10/site-packages/umodbus/client/tcp.py", line 221, in write_multiple_registers
    function.values = values
  File "/usr/local/lib/python3.10/site-packages/umodbus/functions.py", line 1517, in values
    raise IllegalDataValueError
umodbus.exceptions.IllegalDataValueError:  The value contained in the request data field is not an allowable value
    for the server.`

elupus, 2022-10-19:
@dkmh Thanks. Yea we seem to have an encoding issue for writes. Will have to figure out a solution for it.

elupus, 2022-10-19:
@dkmh if you have the ability, feel free to test out: https://github.com/yozik04/nibe/pull/54 i'm i no way sure it is correct thou. Late night here too.

cybermaus, 2022-10-20:
@elupus So I resend the latest `nibe_heatpump` component into `custom_components` to test. Same error.

And this does not fully surprise me, because I did eventually find the function mentioned by @dkmh, but it is in `/nibe/connection/modbus.py`  so not part of what I am dropping into my live HAOS. 

So my method is flawed? I am unsure, because the HP JSON register files are also not in the component, and it seems to find those, so it must be pulling those from somewhere.

To restate my method, that I found somewhere on the net:  I download the entire PR branch as a ZIP, use 7Zip to only get the `components/nibe_heatpump` in a separate ZIP and transfer that to `config/custom_components`  only my normal HAOS box. I do not have any cloned GIT repository and am not installing a separate development HA instance.

elupus, 2022-10-20:
The manifest file was changed... So you need to use the new one and add version again.

elupus, 2022-10-20:
Different issue. If you change your manifest file where it says nibe==1.1.0 to nibe==1.1.1 and try again.

cybermaus, 2022-10-20:
Sorry guys, I have a real customer with an urgent issue. It may be a while.

cybermaus, 2022-10-20:
Latest log:

```
MARKER
2022-10-20 15:19:10.494 ERROR (MainThread) [custom_components.nibe_heatpump] Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/nibe_heatpump/config_flow.py", line 191, in async_step_modbus
    title, data = await validate_modbus_input(self.hass, user_input)
  File "/config/custom_components/nibe_heatpump/config_flow.py", line 144, in validate_modbus_input
    coil = await connection.read_coil(coil)
  File "/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py", line 110, in read_coil
    coil.value = self.coil_encoder.decode(coil, encode_u16_list(result))
  File "/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py", line 45, in encode_u16_list
    return bytes(
  File "/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py", line 46, in <genexpr>
    byte for val in data for byte in val.to_bytes(2, "little", signed=False)
AttributeError: 'numpy.uint16' object has no attribute 'to_bytes'
2022-10-20 15:19:13.219 WARNING (MainThread) [homeassistant.components.rflink] Disconnected from Rflink, reconnecting
```

elupus, 2022-10-20:
@cybermaus if you can hack the path mentioned in your log above, you can change `val.to_bytes` to `int(val).to_bytes` should solve that issue.

yozik04, 2022-10-20:
Main home assistant is separate from SSH container where you are logged in. So it is not possible to edit HA container from SSH Container.

cybermaus, 2022-10-20:
Makes sense. Web UI terminal has the same. How to break this jail?

yozik04, 2022-10-20:
The same way as you would login from one VM to an other 3rd party VM (not owned by you) on the same server. You can't. =)
But it is possible to attach screen and keyboard to the RPI and use terminal in the host OS. From that you would be able to attach to a docker container.

cybermaus, 2022-10-20:
Ok. That is a hassle as I need to physically retrieve the device, but I will do so. If only because I need to get more on top of this docker stuff. But not now, another customer meeting at the top of the hour.

yozik04, 2022-10-20:
you do not need any KVM or VNC. Just use regular raspbian and install docker there. You will have proper access to the host OS. HomeAssistant will run in docker. But this way you will loose opportunity to easy install hass os plugins, of course if you use them. All this requires some linux administration skills though, and maybe some more time and effort.

elupus, 2022-10-20:
You can get into hassos container too: https://community.home-assistant.io/t/protip-how-to-get-shell-in-actual-homeassistant-or-addon-container-when-using-hassio/48760 but it's trigger and you risk breaking stuff.

elupus, 2022-10-20:
I still don't understand thus. umodbus does not seem to be using numpy, so where the heck is that numpy.int16 coming from.

elupus, 2022-10-20:
@cybermaus, change the requirement line in your manifest file to this:
```
"requirements": ["git+https://github.com/yozik04/nibe.git@master#nibe==2.0.0"],
```

The 2.0.0 need to be changed anytime we make some change to nibe library. It's just a dummy version.

elupus, 2022-10-20:
It didn't take the new requirement sadly. Not sure why.

nkrepo, 2022-10-20:
@cybermaus can you post your config.yaml for the SMOS40 so I can join testing ...
I am trying to poll `current-outdoor-temperature-bt1-30002`

```
nibe-mqtt -c nibe-config.yaml
2022-10-20 20:53:41,551 - INFO     - MainThread - root - Running Nibe MQTT 0.3.0 service with Nibe 1.1.1 lib
2022-10-20 20:53:46,563 - INFO     - MainThread - nibe.nibe_mqtt.service - Polling coil current-outdoor-temperature-bt1-30002: last_update = None
2022-10-20 20:53:46,564 - DEBUG    - MainThread - nibe.nibe.connection.modbus - Sending read request
2022-10-20 20:53:46,565 - DEBUG    - MainThread - sockio.TCP(192.168.2.18:502) - open connection (#1)
2022-10-20 20:53:46,576 - INFO     - MainThread - nibe.nibe.connection.modbus - current-outdoor-temperature-bt1-30002: 13.4
2022-10-20 20:53:46,576 - ERROR    - MainThread - nibe.nibe.event_server - 'nibegw'
Traceback (most recent call last):
  File "/home/nkef/.local/lib/python3.10/site-packages/nibe/event_server.py", line 17, in notify_event_listeners
    listener(*args, **kwargs)
  File "/home/nkef/.local/lib/python3.10/site-packages/nibe_mqtt/service.py", line 120, in on_coil_update
    self.mqtt_client.publish_discovery(coil, self._get_device_info())
  File "/home/nkef/.local/lib/python3.10/site-packages/nibe_mqtt/service.py", line 67, in _get_device_info
    "id": slugify("Nibe " + self.conf["nibe"]["nibegw"]["ip"]),
KeyError: 'nibegw'
```

My config is 
```
logging:
  level: DEBUG
  format: "%(asctime)s - %(levelname)-8s - %(threadName)-10s - %(name)s - %(message)s"
mqtt:
  host: 192.168.2.17:1883
  protocol: 5
  username: test
  password: user
nibe:
  model: SMOS40
  word_swap: true
  modbus:
    url: tcp://192.168.2.18:502
    slave_id: 1
  poll:
    coils:
      - current-outdoor-temperature-bt1-30002
```

cybermaus, 2022-10-20:
@nkrepo For what elupus is doing, there is nothing in my configuration.yaml. And I am not using MQTT to the NIBE, but the home assistant MODBUS integration. I dropped my [config here for you already.](https://github.com/yozik04/nibe/issues/32#issuecomment-1279947286) But to restate, that has nothing to do with either MQTT nor the work done here. You are welcome to use it though, but if you have questions about it, best put them in that other issue.

And if you want to start testing the solution of this issue, best follow the instructions from the top. Which somewhat unfortunately start also in the [middle of another issue.](https://github.com/yozik04/nibe/issues/45#issuecomment-1283646518)

nkrepo, 2022-10-20:
> @nkrepo For what elupus is doing, there is nothing in my configuration.yaml. And I am not using MQTT to the NIBE, but the home assistant MODBUS integration. I dropped my [config here for you [...]

@cybermaus 
Sorry I forgot that you already posted the config. 
It fails on the nibe library part so I think the issues are common, but you are correct I should try also through home assistant. 

@elupus 
I would like to join testing,  I can setup anything , an instance of home assistant running in docker , fully supervised, whatever is easier to pull the latest nibe support.

cybermaus, 2022-10-20:
If you can tell me what modbus.py to modify in my root console, I could do that, so you may avoid unneeded versions. Or should I modify all of them?
```
# find / | grep modbus.py | grep nibe
/mnt/data/docker/overlay2/2d19273231b2325c66890146d38205de880645d2fb8c429e53356d5d937056d9/diff/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py
/mnt/data/docker/overlay2/c221e9b9939e25592f42b86f9b204322db2820610b0312846ff6ebba8b484778/diff/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py
/mnt/data/docker/overlay2/c221e9b9939e25592f42b86f9b204322db2820610b0312846ff6ebba8b484778/merged/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py
/var/lib/docker/overlay2/2d19273231b2325c66890146d38205de880645d2fb8c429e53356d5d937056d9/diff/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py
/var/lib/docker/overlay2/c221e9b9939e25592f42b86f9b204322db2820610b0312846ff6ebba8b484778/diff/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py
/var/lib/docker/overlay2/c221e9b9939e25592f42b86f9b204322db2820610b0312846ff6ebba8b484778/merged/usr/local/lib/python3.10/site-packages/nibe/connection/modbus.py
```

elupus, 2022-10-20:
I've published a new lib and updaded the pull request.

@cybermaus  change to `"requirements": ["nibe==1.1.2"],`

elupus, 2022-10-20:
ps. @cybermaus you should not be modifying those. You should do as in: https://community.home-assistant.io/t/protip-how-to-get-shell-in-actual-homeassistant-or-addon-container-when-using-hassio/48760 and enter the docker container. Then you can find the paths in the error logs. But.. be careful :)

dkmh, 2022-10-20:
No success here in pulling nibe requirement directly from github master instead of pypi release. No idea why.

But with version 1.1.2:
[image]
(Changing value works too 👍 )
[image]
🥳 Many THX for your work so far, will start further testing tomorrow.

cybermaus, 2022-10-20:
It is absolutely marvellous.
Not wanting to do too much more today, I figured I should at least enable the ones I was already monitoring, so I have some history to look at tomorrow.

Which leads me to the fact that this one seems to be missing:
```
#Priority	MODBUS_INPUT_REGISTER	1028	1		4	0	0	0
# 10: Off, 20: Hot Water, 30: Heat, 40: Pool, 50: Cooling
```

elupus, 2022-10-20:
Hehe, yea then exports seem unreliable. I have opened #56 and #52

closed 2022-10-20 by elupus