yozik04/nibe #77: SMO40 - Filter time: wrong format

tomrennen, 2022-12-19:
I'm using the HA core library made by @elupus. I noticed it exposes the `Outdoor filter time` parameter (`47377`). This is exposed as being an enum with either 12 hours or 24 hours selection. This setting however in reality allows any value between 1-48, meaning this currently gets mapped to `Unknown` if it's not set to either 12 or 24. Is this as easy as updating the respective lines in [the data json](https://github.com/yozik04/nibe/blob/master/nibe/data/smo40.json#L7393)? Or should the HA Core component handle these types of exceptions to the exports?

elupus, 2022-12-19:
It should be fixed in the data file yes, using the extension file.

tomrennen, 2022-12-19:
I'm not sure how to remove a value using the extensions file, would this be `mappings: {}` or `mappings: null`?

elupus, 2022-12-19:
You should map 12 to "12 HOURS" and 24 to "24 HOURS"

tomrennen, 2022-12-19:
I'm not sure you've read my first post correctly. It is actually an enum now, which allow selection of `12 HOURS` and `24 HOURS`, but I think this is wrong. The allowed values are `1-48`, not just `12` and `24`. Mine is currently set to `4`, which is rendered to `Unknown` in Home assistant, as there's no case for `4 HOURS`. I'd prefer this field to be of type `number` instead of `select`, meaning we'll have to remove the enum cases right? Would `mappings: {}` or `mappings: null` in extensions achieve the desired result?

elupus, 2022-12-19:
Ah, removing a mapping i dont know if its supported. But mapping:  null seem an ok way to indicate it. Needs fixes in the convert script to handle it too.

elupus, 2022-12-22:
https://github.com/home-assistant/core/pull/84449

tomrennen, 2022-12-28:
It is indeed a slider now in current beta, thank you @elupus!