yozik04/nibe #83: Separate coil configuration from data

elupus, 2023-01-01:
Right now the coil object database ends up being state-full, with the value being stored in the object loaded by the json database. This make things somewhat tricky to test and can have odd behaviours with the coil value changing behind ones back.

```python
coil = heatpump.get_coil(1234)
coil.value = 1

await asyncio.sleep()
# During the await, we get an out of band update of coil 1234,
# which will revert it's value to what it was before we changes it's value to 1, let's say 0.

# Now we try to write the coil, which have reverted back to 0.
await connection.write_coil(coil)
```

The coil value should not be stored in the Coil object.

yozik04, 2023-01-01:
I was initially thinking about it. But decided to have value inside coil. What is wrong with testing?

elupus, 2023-01-01:
doing stuff like write_coil.assert_called_with() is hard, since you need to be able to provide a new instance of a Coil object with same parameters as the original Coil object in the data base, but with the expected value.

If you naively grab the Coil from the coil db, and update it's value to the expected value, you break the instance that is you are checking against.

yozik04, 2023-02-06:
I was thinking how to approach that. Decided to create such a class.

```python3
@dataclass
class CoilData:
    coil: Coil
    value: Union[int, float, str, None] = None

# def is_valid:
# ... etc ...
```

closed 2023-02-28 by yozik04