yozik04/nibe #84: VVM S320 registers missing

haerq, 2023-01-18:
I exported my registers from newly installed VVM S320 R and compared it to S320_S235.csv.
It seems to be 76 missing registers.

Haven't used GitHub much so I didn't know how I should add those registers, but here is attached csv which seems to include all previous registersand also 76 new.
Maybe it's possible to add those registers?
[modbus_addresses_all_20230118.csv](https://github.com/yozik04/nibe/files/10452021/modbus_addresses_all_20230118.csv)

yozik04, 2023-01-30:
Do you have latest firmware installed?

haerq, 2023-01-30:
> Do you have latest firmware installed?

Yes I have. It's 2.16.10 and there is no update available in myUplink.

steffen195853, 2024-01-02:
HI, there are continous firmware upgrades happening for "my" VVM-S320. How do you follow the changes of Modbus registers coming with this upgrades? Is it helpful tho share again a latest export of the Modbus registers for firmware 2.23.4 which is the latest public available version ?

yozik04, 2024-01-02:
You can make an export and compare with what we have in data folder. If changes are useful then you can share the file with us.