yozik04/nibe #87: TCP Modbus example does not work?

hansij66, 2023-02-05:
Windows10
Python 3.9
PyCharm

I just got a SMOS40/S2125 installed. I was able to integrate with Home Assistant and get some readings (hence, TCP MODBUS is working).

If I use the README.md TCP Modbus example, I do get errors...and in this example a bit clueless what to change. I literately took README.md, changed to Model.SMOS40 and try to read current-outdoor-temperature-bt1-30002.

I do get the following errors in PyCharm/Windows (I reduced paths a bit to make readable). Any hints appreciated! :) Thanks
```
DEBUG:asyncio:Using proactor: IocpProactor
DEBUG:nibe.nibe.connection.modbus:Sending read request
DEBUG:sockio.TCP(192.168.1.16:502):open connection (#1)
Traceback (most recent call last):
  File "nibe-s2125\test.py", line 31, in <module>
    loop.run_until_complete(main())
  File "Python39\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "nibe-s2125\test.py", line 23, in main
    await connection.read_coil(coil)
  File "Python39\lib\site-packages\nibe\connection\modbus.py", line 86, in read_coil
    result = await self._client.read_input_registers(
  File "Python39\lib\site-packages\async_modbus\core.py", line 194, in read_input_registers
    return await self._send_message(request)
  File "Python39\lib\site-packages\async_modbus\core.py", line 142, in _send_message
    return await self.protocol._async_send_message(
  File "Python39\lib\site-packages\async_modbus\core.py", line 39, in send_message_tcp
    await writer.drain()
  File "Python39\lib\site-packages\async_modbus\core.py", line 113, in drain
    await self._write_coro
  File "Python39\lib\site-packages\sockio\aio.py", line 31, in wrapper
    await self.open()
  File "Python39\lib\site-packages\sockio\aio.py", line 287, in open
    self.reader, self.writer = await coro
  File "Python39\lib\site-packages\sockio\aio.py", line 158, in open_connection
    configure_socket(sock, no_delay=no_delay, tos=tos, keep_alive=keep_alive)
  File "Python39\lib\site-packages\sockio\aio.py", line 117, in configure_socket
    sock.setsockopt(socket.SOL_IP, socket.IP_TOS, tos)
  File "Python39\lib\asyncio\trsock.py", line 82, in setsockopt
    self._sock.setsockopt(*args, **kwargs)
OSError: [WinError 10022] An invalid argument was supplied
```

elupus, 2023-02-05:
I dont think it work on windows. Its not been a target platform. It does seem to be an issue in the sockio library.

elupus, 2023-02-05:
Some other that mention issues with that lib on wondows: https://github.com/tiagocoutinho/sockio/issues/5

You can test in wsl2 on windows instead.

hansij66, 2023-02-06:
I will try Linux...that will be the target anyway...but my PyCharm is on a
Windows10 laptop :) That is fixable. Thanks, will test this week.

closed 2023-02-13 by hansij66

bj00rn, 2025-05-07:
Does not work in MacOS either. python 3.11