yozik04/nibe #90: SMO S40 no longer seen after full reset - custom integration

cybermaus, 2023-02-16:
So back in November, I was amoung the ones to beta test the SMO S40, using custom Home Assistant integration.
It was always working, but now after a full factory reset on my SMO S40, I lost all connection.

I did remember to reactivate the Modbus TCP , and in fact I can actually read it usign the Homo Assistant direct modbus integration.

Obvious choice is to delete the device, and re-add it from scratch.
But I am thinking, I probably also want to do that with the. surely by now no longer custom integration.
What is the best way to remove the custom integration from November?

elupus, 2023-02-16:
Just remove the nibe_heatpump folder from inside your config/custom_components folder.

cybermaus, 2023-02-16:
Ok, works, thanks. After first removing, and then adding the new one (rather then duplicating de sensors) it even picked up the history correctly:

[image]

closed 2023-02-16 by cybermaus