yozik04/nibe #103: Support for S735 wanted

amattsso, 2023-05-03:
Hi,

Recently installed a Nibe S735 heat pump. I'd very much would like this heat pump to be supported thru the native modbus integration.

Attaching the registerdump. Unfortunately in Swedish. Please respond if you have to have this in English. i suppose I could switch the language on the pump to english temporarily, if that would be of any help.

[modbus_addresses_all_20230502-06613223019004-1.csv](https://github.com/yozik04/nibe/files/11379918/modbus_addresses_all_20230502-06613223019004-1.csv)

yozik04, 2023-05-03:
Yes. Please switch to English then do the export. Thank you.

amattsso, 2023-05-03:
Hi,

Here it comes :).

[modbus_addresses_all_20230503-06613223019004-1.csv](https://github.com/yozik04/nibe/files/11381457/modbus_addresses_all_20230503-06613223019004-1.csv)

amattsso, 2023-05-03:
Yes, I can test if you can give me some information/guidelines.
Haven't done it before.

yozik04, 2023-05-03:
Depends on your experience with IT. If you feel confident with git, can install a project with pip and is comfortable running tools in command line then we can try.

amattsso, 2023-05-03:
Hi,

I'm fairly experienced in IT generally, but a newbee in git, linux and HA
beneath the GUI...
I tried installing and configuring the Git Pull Add.on but failed on the
repository config.
I'm happy to follow your best practice setup.

yozik04, 2023-05-03:
If you have python installed you can try to run `pip3 install nibe`. That should install command line tool called: `nibe-cli`. The tool has invocation help built in. Feel free to try. I have released `2.2.0`. @elupus needs to update HomeAssistant integration for the new lib version.

amattsso, 2023-05-03:
Hi,

I'm running a very standard HA (Home Assistant OS 10.1, Core 2023.4.6 and
have not actively installed python in the environment.
So if it is not part of the core there will be some additional steps I
assume.
I tried the command using Terminal & SSH Add-on but that failed.
Any ideas?

yozik04, 2023-05-04:
=) Then, discard all the above and wait for a new integration version.

yozik04, 2023-05-04:
Thank you again for the export.

closed 2023-05-04 by yozik04

amattsso, 2023-05-05:
Hi,

Sorry I could not test the work you put down.
I guess I had too little knowledge in the process :(
I'll wait for the release whenever that is ready, and give it a test then.
Do you know if it will make it to the 2023.6 reales?
Thanks for your fast replies :).