yozik04/nibe #105: Add support for cooling curve parameters for S series (SMOS40)

werfpsa, 2023-05-20:
Can you please add the cooling parameters for the SMOS40 (curve, etc)

[mobus_all_adresses_2023_04_25.xlsx](https://github.com/yozik04/nibe/files/11521878/mobus_all_adresses_2023_04_25.xlsx)

Min. supply temp. cooling climate system 1	MODBUS_HOLDING_REGISTER	720	1	Â°C	1	7	30	18
Own curve, cooling P3	MODBUS_HOLDING_REGISTER	721	1	Â°C	1	7	40	20
Own curve, cooling P5	MODBUS_HOLDING_REGISTER	722	1	Â°C	1	7	40	20
Cooling connected, climate system 1	MODBUS_HOLDING_REGISTER	726	1		4	0	1	1
Period time cooling	MODBUS_HOLDING_REGISTER	736	1	min	4	0	180	30
Cooling delta temp. 20Â°C	MODBUS_HOLDING_REGISTER	738	1	Â°C	1	3	10	3
Cooling delta temp. 40Â°C	MODBUS_HOLDING_REGISTER	739	1	Â°C	1	3	20	6
Operating mode circulation pump cooling heat pump 2	MODBUS_HOLDING_REGISTER	831	1		4	0	1	1
Operating mode circulation pump cooling heat pump 1	MODBUS_HOLDING_REGISTER	832	1		4	0	1	1
(SPA), cooling activated	MODBUS_HOLDING_REGISTER	849	1		4	0	1	1
Cooling curve climate system 1	MODBUS_HOLDING_REGISTER	967	1		1	0	9	0
Cooling offset climate system 1	MODBUS_HOLDING_REGISTER	975	1		1	-10	10	0
Own curve, cooling P4	MODBUS_HOLDING_REGISTER	976	1	Â°C	1	7	40	20
Own curve, cooling P2	MODBUS_HOLDING_REGISTER	977	1	Â°C	1	7	40	20
Own curve, cooling P1	MODBUS_HOLDING_REGISTER	978	1	Â°C	1	7	40	20

elupus, 2023-05-20:
Please provide data in original csv format.

werfpsa, 2023-05-20:
[modbus_addresses_all_20230520-06765421302061-1.csv](https://github.com/yozik04/nibe/files/11522206/modbus_addresses_all_20230520-06765421302061-1.csv)
Here you go, sorry for this.

yozik04, 2023-05-20:
I have added all new that I found.

closed 2023-05-20 by yozik04

werfpsa, 2023-05-20:
Super and thanks