yozik04/nibe #111: F750 Airflow sensors lack of units

pacco788, 2023-07-25:
Hi, I'm using the NIBE Integration in HomeAssistant. Was wondering why there are no units for the airflow sensors. Couldn't use it for automations and it didn't show a graph. 

After a little research I found the problem. The airflow sensors in your F750 file have no units. Should be m3/h. Affects EB100-BS1 Air flow (40050), Airflow ref. (43124) and EB100-BS1 Air flow unfiltered (40051).

Would be nice if you could add it.

pacco788, 2023-07-25:
Ouch, didn't see that. 40051 wasn't in use in my setup. Activated it for now.

The value of EB100-BS1 Air flow (40050) in HA is the same as in the NIBE display (m3/h), that's correct. 40051 is a tenth of 40050. I could solve this in HA for myself, but I don't know how to suggest a value for that. m3/h/10 sounds not very charming ;-)

tizianodeg, 2024-06-14:
Greate there is an issue opened!
@yozik04 not sure how you can make sure that values are correct. 
But at least there is a need to override values from the modbus manager. 
Beside of the missing units for 40050 and 40051 and the factor, there is also wrong units for fan_speed_current_43108
it is shown as % but the value is the selected Fan mode e.g. a value between 0-4
@pacco788  is that the same for you

elupus, 2024-06-15:
You need to fix this in https://github.com/yozik04/nibe project.

Mainly by adjusting the extension filer here: https://github.com/yozik04/nibe/blob/master/nibe/data/extensions.json and running the generation script (https://github.com/yozik04/nibe/blob/master/README.md#2-convert-source-csv-files-to-json)

tizianodeg, 2024-06-15:
Thank you @elupus reading the doc more closely may have helped me ;)

I will fix:
sensor.eb100_adjusted_bs1_air_flow_41026
sensor.eb100_bs1_air_flow_40050
sensor.eb100_bs1_air_flow_unfiltered_40051

all of them affects only F750 and f730. Having a f730 I can test it:
[image]

For the Fan speed current I will create a separate request...
As they are used by different Models and I'm unsure if all have the same issue...
Do you have tester for:
f370_f470
f1145_f1245
f1155_f1255
f1345
f1355

For my F730 this makes more sense:
[image]
than:
[image]

My Nibe app however shows 30% - 30% is configured for Mode 2 on my system.

yozik04, 2024-06-16:
Do you have the last version of the firmware?

I thought that all F1xxx series are ground heat pumps. No idea if these can have any fan unit attached.

tizianodeg, 2024-06-16:
I just checked where the id is used:
https://github.com/search?q=repo%3Ayozik04%2Fnibe+41258&type=code
Guess it depends if a ventilation is installed or not like a Exhaust air module FLM?

No idea I can only test my F730, could do the change also only for the f730/F750 guess they should behave similar. 
Let move the discussion on current fan speed to #164. 
For the Air flow I created a PR it only affects F750 and F730 as they are the only one with those parameters.

closed 2025-02-22 by yozik04