yozik04/nibe #116: Out of range values from some pumps

elupus, 2023-10-06:
Seems some pumps have really weird out of range values. See https://github.com/home-assistant/core/issues/100852#issuecomment-1751118757 these cause reading to throw exceptions.

I'm pondering if we should demote range checks for decode to only log warnings, or even just log debug messages. We can still validate on write.

yozik04, 2023-10-07:
If you read a single coil then `ReadException` is what you expect on failure. When a full table is read then, I think, it just logs a warning.

elupus, 2023-10-07:
Well, its not like the max/min have a real affect on things. We can still represent the value. So getting the out of range value (on read) might be more useful, given the rather poor quality of the databases of parameters.

Raising on write does make sense, to not confuse the pump.

yozik04, 2023-10-07:
I've seen invalid temperature readings previously. My MAX485 module is not that good because I see some mess on the line. After adding pull up registers oscilloscope started to display a better picture but far from perfect.. I do get read errors quite frequently. I was thinking that in the read response packet checksum somehow managed to be correct but readings were off (1000 degrees or above).

Also during a reboot you can get out of range values quite easily. Maybe we fixed it with 
[def _is_hitting_integer_limit(self, coil: Coil, int_value: int):](https://github.com/yozik04/nibe/blob/1715cee43541e409ca288f42c32e542c16243273/nibe/connection/encoders.py#L78-L92)

I'd better keep range checks to keep these two issues away.

closed 2025-02-22 by yozik04