yozik04/nibe #131: Decoder error

zlorf, 2023-11-23:
I've recently updated from 2.4.0 to 2.5.0 because I've seend that modbus s8 handling has been fixed.

However, I've got a following exception:
```
Traceback (most recent call last):
  File "/home/jacek/dev/smarthome/nibe/nibe_monitor2.py", line 207, in <module>
    loop.run_until_complete(main())
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/jacek/dev/smarthome/nibe/nibe_monitor2.py", line 88, in main
    data[name] = await connection.read_coil(coil)
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/_asyncio.py", line 88, in async_wrapped
    return await fn(*args, **kwargs)
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/_asyncio.py", line 47, in __call__
    do = self.iter(retry_state=retry_state)
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/__init__.py", line 314, in iter
    return fut.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 433, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/_asyncio.py", line 50, in __call__
    result = await fn(*args, **kwargs)
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/modbus.py", line 111, in read_coil
    coil_data = self.coil_encoder.decode(coil, result)
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/encoders.py", line 84, in decode
    value = self.decode_raw_value(coil.size, raw)
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/encoders.py", line 166, in decode_raw_value
    raw_bytes = [byte for value in raw for byte in value.to_bytes(2, "little")]
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/encoders.py", line 166, in <listcomp>
    raw_bytes = [byte for value in raw for byte in value.to_bytes(2, "little")]
AttributeError: 'numpy.uint16' object has no attribute 'to_bytes'
```

Python 3.9.2

Seems that some library returned numpy numbers instead of ints (?).

elupus, 2023-11-23:
Annoying!!! :). Will have to fix that, thanks for the heads up. Need to fix before next HA release.

zlorf, 2023-11-23:
Thank you for the recent fix in the first place!
I had a workaround previously where I replaced all 's8' types with 's16' to be able to write negative values, but it would be nice to get rid of this hack :)

zlorf, 2023-11-23:
From https://pypi.org/project/async-modbus/:

> Numpy will be used if installed.

closed 2023-11-23 by yozik04

zlorf, 2023-11-23:
Thanks for the quick fix! I confirm it works not with s8 properly :)

PR #132 (2023-11-23): Allow numpy int16 in decode
Sometimes umodbus returns a numpy int16 instead of real python ints, so let the decode function explicitly only require an int:able type and wrap it in an int conversion.

Fixes: #131