yozik04/nibe #133: Modbus problem writing negative degree-minutes

zlorf, 2023-11-23:
It was a problem on 2.4, it's still not working on 2.5.1.

Python 3.9.2, Nibe S320. Tested on 2.4 and 2.5.1

Whenever I try to write degreeminutes field (which is usually negative), it gets stored as 0.
I did some deep-down debugging and it was really weird, as the bytes were similar. If I tried to write what I've received from modbus, the library send the same stream of bytes, and yet value got reset to 0.

I then discovered (by trials and errors, honestly), that if I reverse the list of registers:
```
connection = Modbus(heatpump=heatpump, url="tcp://" + NIBE_IP + ":502", slave_id=1)
# Patch write method to work with degreeminutes
orig = connection._client.write_registers
connection._client.write_registers = lambda slave_id, starting_address, values: orig(slave_id, starting_address, list(reversed(values)))
```
then it works... as long as I stay on the same side of 0. That is: if DM are already negative, I can write any negative number correctly. But if they are positive, then writing negative number results in 0. I haven't tried writing positive number when DM is negative.

I really wonder if this is maybe a fault with my heatpump.
But maybe you have an idea what is wrong. s32 type is little tricky. No rush, as the workaround does the trick for me.

elupus, 2023-11-24:
Since im curious. Can you test to see if behaviour changes depending on if python numpy is installed?

elupus, 2023-11-24:
Oh right you say you art the network trace. If you read and write same raw data in the trace it really should work. So then it does seem like a pump bug.

closed 2025-02-22 by yozik04