yozik04/nibe #134: Coil read error

zlorf, 2023-11-23:
Nibe 2.5.1
Unfortunatelly, I do not know which coil caused the error.

```
Traceback (most recent call last):
  File "/home/jacek/dev/smarthome/nibe/nibe_monitor.py", line 215, in <module>
    loop.run_until_complete(main())
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/jacek/dev/smarthome/nibe/nibe_monitor.py", line 99, in main
    data[name] = await connection.read_coil(coil)
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/_asyncio.py", line 88, in async_wrapped
    return await fn(*args, **kwargs)
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/_asyncio.py", line 47, in __call__
    do = self.iter(retry_state=retry_state)
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/__init__.py", line 314, in iter
    return fut.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 433, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/jacek/.local/lib/python3.9/site-packages/tenacity/_asyncio.py", line 50, in __call__
    result = await fn(*args, **kwargs)
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/modbus.py", line 111, in read_coil
    coil_data = self.coil_encoder.decode(coil, result)
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/encoders.py", line 85, in decode
    if self._is_hitting_integer_limit(coil.size, value):
  File "/home/jacek/.local/lib/python3.9/site-packages/nibe/connection/encoders.py", line 99, in _is_hitting_integer_limit
    return int_value >= limit
TypeError: '>=' not supported between instances of 'NoneType' and 'int'
```

elupus, 2023-11-24:
That was strange indeed. That would have been caught by our tests.

elupus, 2023-11-24:
Can you test https://github.com/yozik04/nibe/pull/135?

zlorf, 2023-11-24:
Thanks for a quick fix; it seems to be able to address the issue.
Unfortunately I cannot check it today. :(

PR #135 (2023-11-25): Corrected doubled checks for limit on modbus
Corrected doubled checks for limit on modbus

Fixes: #134
commit b05c292b (2023-11-25): Corrected doubled checks for limit on modbus (#135)

Corrected doubled checks for limit on modbus

Fixes: #134
closed 2023-11-25 by elupus