yozik04/nibe #138: Add support for external energy meter BE6

Freiheiter, 2023-12-10:
Got the external pulse energy meter (based on the S0 interface) to work with my Nice S1255-6 PC over modbusTCP and thought I'd share how.

I'm aware that my setting is against the spec, which claims the register to be 398 for BE6, and with an offset of 30001 we should end up at 30399. But the USB-extract retrieved from the pump gives 396, and the respective 30397 works like a charm. So I suspect the spec is in error here. Please consider for a merge.

Here's what I added to s1155_s1255.json:

"30397": {
"title": "Pulse energy meter (BE6/BF2)",
"factor": 100,
"size": "u32",
"unit": "kWh",
"name": "pulse-energy-meter-be6-bf2-30397"
},

Andy-Voigt, 2024-07-04:
Same for the s320

yozik04, 2024-07-04:
@Andy-Voigt can you please temporarily change heat pump language to English and then do the export.

Andy-Voigt, 2024-07-04:
[modbus_addresses_all_20240704-06920621343032-1.csv](https://github.com/user-attachments/files/16095569/modbus_addresses_all_20240704-06920621343032-1.csv)

Here is the csv in English.

yozik04, 2024-07-04:
I will. Thank you.

closed 2024-07-04 by yozik04

Andy-Voigt, 2024-07-04:
Is it realy a read only register?

Freiheiter, 2024-07-04:
By the specs, it is read-only.
The value is calculated from pulses received from an external contact, typically connected to an electricity meter, using the protocol S0 as in EN 62053-31
It might be helpful to be able to overwrite that values (for example, if it diverges from the meter's physical display for whatever reason), but by the specs, this can't be done.

PR #167 (2024-07-04): s320 and s325 registers update
Resolves #138
commit 034a3b9f (2024-07-04): s320 and s325 registers update (#167)

Resolves #138