yozik04/nibe #140: Hot water demand stopped working after upgrade to 2.5.0 after switching to new selector (replacing number)

malatg, 2023-12-14:
[image]

elupus, 2023-12-14:
Do you have a log?

malatg, 2023-12-14:
I have activated the debug log but there is nothing in the logs from the nibe integration.

malatg, 2023-12-14:
Hi,

Yes this is what I get

2023-12-14 22:50:56.806 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for hot-water-demand-mode-40057)
2023-12-14 22:50:56.806 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 0.181 seconds (success: True)
2023-12-14 22:51:56.805 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for hot-water-demand-mode-40057)
2023-12-14 22:51:56.805 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 0.180 seconds (success: True)

elupus, 2023-12-15:
Can you turn on debug logs for the integration and see if you get some more info?

malatg, 2023-12-15:
Sorry tried that but couldn't find any more info (it still ends up under "home assistant core" log right?)

elupus, 2024-10-16:
Closing since i've not received any updates on this.

closed 2024-10-16 by elupus

malatg, 2024-10-16:
Sorry for not updating it solved itself a couple of upgrades ago👍