yozik04/nibe #148: Incorrect registers for weather control data (S1155)

lewurm, 2024-01-21:
I've noticed that any values regarding weather forecast are always null:
https://github.com/yozik04/nibe/blob/9dde26cccb04cc3b73f5a6349b3d53869c65c56e/nibe/data/s1155_s1255.csv#L500-L503

While in myuplink something like this is shown:
[image]

Does anyone get useful weather data?

yozik04, 2024-01-22:
Interesting what they are using it for. Forecast for just 6 hours ahead does not give much if you have floor heating.
Check what registers are read in MyUplink.

lewurm, 2024-01-22:
> Interesting what they are using it for. Forecast for just 6 hours ahead does not give much if you have floor heating.

My understanding is that this is an improvment over controlling the heating power by the average outdoor temperature, as it's reacting better to drastic weather changes. But indeed the available information on this feature is quite modest.

I'd like to understand it better, that's why I started to poke at this.  For the SPA (Smart Price Adaption) feature there is a `Heating offset (SPA)` register which shows what is happening behind the scenes, e.g. from last week:
[image]

There is a clear relationship between electricity price, this computed offset and the compressor frequency.

Ideally I would like to have something in similar vain for the weather feature, e.g. `Heating offset (weather)`.

--------------------------------------------------------------------

> Check what registers are read in MyUplink.

Interesting, I didn't know that modbus specs do show up when using MyUplink.  Alas for those fields `MDOBUS_NO_REGISTER` is used as `modbusRegisterType` 🙁 
[image]

So nothing can be done, or is there more to it?

yozik04, 2025-02-22:
If it is MODBUS_NO_REGISTER, then it is probably fetched not via MODBUS. Closing. I think out registers are correct.

closed 2025-02-22 by yozik04