yozik04/nibe #159: erroneous modbus values eb100-beX - NibeGW ESPHOME - F1255

MichaelPihlblad, 2024-05-04:
Getting these erroneous values in homeassistant:

```
INFO (MainThread) [nibe.nibe.connection.nibegw] Coil eb100-be3-current-40079, value: 13107.2
INFO (MainThread) [nibe.nibe.connection.nibegw] Coil eb100-be2-current-40081, value: 65536.0
INFO (MainThread) [nibe.nibe.connection.nibegw] Coil eb100-be1-current-40083, value: 32768.0
```

Checked the csv and json files for F1255 and they are correct.

It used to work with the same heat pump while using NibePI. So most likely not the heatpump raw values that are incorrect.

elupus, 2024-05-04:
Change your wordswap setting

MichaelPihlblad, 2024-05-04:
Thanks, I was thinking it could possibly be byte order issue to look into in this lib.

But there is an option in the heat pump to swap the byte order, available over modbus even, and that solves the issue.

closed 2024-10-12 by elupus