yozik04/nibe #160: Invalid values slip through

yozik04, 2024-05-13:
Unfortunately we have a regression that now invalid values slip into HomeAssistant:

NibeGw, register address 40016 (EB100-EP14-BT11 Brine Out Temp). Which is signed 16bit.
[image]

yozik04, 2024-05-13:
Hmm:
```
Int16sl.parse(b'\x00\x80')
-32768

Int16sl.parse(b'\x01\x80')
-32767
```

yozik04, 2024-05-13:
I see that at the same time other values were also read wrong.
[image]

I think we should add some checks and discard whole received packet if any of the values are not invalid. I have no idea how it manages to bypass checksum check. I see this happen from time to time. Noticed second time this week.

elupus, 2024-05-13:
Looks quite weird. I wonder how we could detect that case.

closed 2025-02-22 by yozik04