yozik04/nibe #168: F1255 holiday-activated-48043 decoding problems

yozik04, 2024-07-17:
I failed to deactivate Holiday mode yesterday. It was certainly working before. It might be related to my HeatPump FW update.

```
Jul 14 11:34:50 smarthomepi python3[992]: 2024-07-14 11:34:50,603 - WARNING  - Ignoring coil 48043 value 01000000 - failed to decode
Jul 14 11:34:50 smarthomepi python3[992]: --- Logging error ---
Jul 14 11:34:50 smarthomepi python3[992]: Traceback (most recent call last):
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/coil.py", line 123, in get_mapping_for
Jul 14 11:34:50 smarthomepi python3[992]:     return self.mappings[str(value)]
Jul 14 11:34:50 smarthomepi python3[992]:            ~~~~~~~~~~~~~^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]: KeyError: '1'
Jul 14 11:34:50 smarthomepi python3[992]: During handling of the above exception, another exception occurred:
Jul 14 11:34:50 smarthomepi python3[992]: Traceback (most recent call last):
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/connection/encoders.py", line 88, in decode
Jul 14 11:34:50 smarthomepi python3[992]:     return CoilData.from_raw_value(coil, value)
Jul 14 11:34:50 smarthomepi python3[992]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/coil.py", line 196, in from_raw_value
Jul 14 11:34:50 smarthomepi python3[992]:     return CoilData.from_mapping(coil, value)
Jul 14 11:34:50 smarthomepi python3[992]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/coil.py", line 181, in from_mapping
Jul 14 11:34:50 smarthomepi python3[992]:     return CoilData(coil, coil.get_mapping_for(value))
Jul 14 11:34:50 smarthomepi python3[992]:                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/coil.py", line 125, in get_mapping_for
Jul 14 11:34:50 smarthomepi python3[992]:     raise NoMappingException(
Jul 14 11:34:50 smarthomepi python3[992]: nibe.exceptions.NoMappingException: Mapping not found for holiday-activated-48043 coil for value: 1
Jul 14 11:34:50 smarthomepi python3[992]: The above exception was the direct cause of the following exception:
Jul 14 11:34:50 smarthomepi python3[992]: Traceback (most recent call last):
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/connection/nibegw.py", line 281, in read_coil
Jul 14 11:34:50 smarthomepi python3[992]:     return await asyncio.wait_for(future, timeout)
Jul 14 11:34:50 smarthomepi python3[992]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]:   File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
Jul 14 11:34:50 smarthomepi python3[992]:     return fut.result()
Jul 14 11:34:50 smarthomepi python3[992]:            ^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/connection/nibegw.py", line 432, in _on_raw_coil_value
Jul 14 11:34:50 smarthomepi python3[992]:     coil_data = self.coil_encoder.decode(coil, raw_value)
Jul 14 11:34:50 smarthomepi python3[992]:                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 14 11:34:50 smarthomepi python3[992]:   File "/smarthome/services/nibe/venv/lib/python3.11/site-packages/nibe/connection/encoders.py", line 91, in decode
Jul 14 11:34:50 smarthomepi python3[992]:     raise DecodeException(
Jul 14 11:34:50 smarthomepi python3[992]: nibe.exceptions.DecodeException: Failed to decode holiday-activated-48043 coil from raw: 01000000, exception: Mapping not found for holiday-activated-48043 coil for value: 1
```

elupus, 2024-07-17:
Have you done the myuplink firmware update?

yozik04, 2024-07-17:
Yes. I did.

elupus, 2024-07-17:
I had a report that the alarm clearing reported negative responses too.

Have you tried other writes?

yozik04, 2024-07-19:
I frequently use temporary-lux-48132 and it works fine.

yozik04, 2024-07-19:
allow-cooling-47372 also works fine.

elupus, 2024-07-19:
Good to hear. So maybe just rejecting the attempt to alarm clear with zero clear. Will just make sure we accept a negative response in HA.

elupus, 2024-07-26:
Did you figure out any solution? Could it be that the valid value have always been 1, but the earlier pumps have accepted 10 as a written value?

yozik04, 2024-07-27:
I have not spent any time on it. Was busy with other things.

I am sure it was 10 before, otherwise I wouldn't be able to see that I triggered it.

I can update Activated to 1 and send it for a scream test :)

yozik04, 2024-07-27:
I am now thinking if we should pass values that failed mapping up to HomeAssistant. Let's assume we do not have a mapping for some read raw value, instead of ignoring it we would pass it to HA like "Unknown (`hex or int`)". If the unit has responded with that value, then it is valid, why we should ignore it then?

Any thoughts on that?

elupus, 2024-07-27:
Agree. Would make it easier to find and correct.

yozik04, 2024-07-27:
Solved here #172

elupus, 2024-10-12:
I've also checked now and my pump is using 1 for holiday mode active so i've changed that here: #178

elupus, 2024-10-12:
Ps. Degree Minutes 32bit is fully funky after the upgrades too.

closed 2025-02-22 by yozik04