yozik04/nibe #180: Heatpump denied writing

yozik04, 2024-10-13:
Strange. I was getting next error messages into the log for every write to allow-heating-47371 (F1255).
Tried manually toggling both sides and got no errors in the log.
Writing to investigate further.

```
Oct 10 10:00:04 pi python3[1719210]: 2024-10-10 10:00:04,334 - ERROR    - Heatpump denied writing allow-heating-47371
Oct 10 22:00:05 pi python3[1719210]: 2024-10-10 22:00:05,619 - ERROR    - Heatpump denied writing allow-heating-47371
```

elupus, 2024-10-13:
There was a report that the firmware update from a few days ago fixed that

yozik04, 2024-10-13:
Will test new firmware then. It started on 2024-09-27 for me... Do not remember updating the firmware.

yozik04, 2024-10-14:
Same thing with 9699R7.
2024-10-14 06:00:04,928 - ERROR    - Heatpump denied writing allow-heating-47371

But it is just a log line, in reality it has changed.

elupus, 2024-10-14:
For alarm it has been confirmed fixed by firmware update just todat: 

https://github.com/home-assistant/core/issues/120023#issuecomment-2411727220

So maybe they just fixed it in one place :)

mattfro, 2024-10-21:
Hmm... well, actually, It's back again, not sure why it was possible to reset alarm ones.

mattfro, 2024-10-21:
Seems in myuplink it's not there(or its in the archive) but in homeassistant the alarm is visible. And I can't reset it getting same "error". Unfortunately can't check what the heatpump itself shows, alarm or not.  It worked fine when resetting it once at least after the firmware update.

yozik04, 2025-02-22:
Will close this. I think it is solved.

closed 2025-02-22 by yozik04

plopp, 2025-10-03:
Got this immediately after when resetting the fan mode to normal. This error seems to be intermittent.
```
DEBUG:nibe.nibe.connection.nibegw:Sending c06b069cb80000000089 (write request) to 192.168.4.210:10001
DEBUG:nibe.nibe.connection.nibegw:Received 5c00206b004b9c from ('192.168.4.210', 10000)
DEBUG:nibe.nibe.connection.nibegw:Container: 
    address = (enum) MODBUS40 32
    cmd = (enum) MODBUS_WRITE_REQ 107
    data = b'' (total 0)
DEBUG:nibe.nibe.connection.nibegw:Received 5c00206b004b9c from ('192.168.4.210', 10000)
DEBUG:nibe.nibe.connection.nibegw:Container: 
    address = (enum) MODBUS40 32
    cmd = (enum) MODBUS_WRITE_REQ 107
    data = b'' (total 0)
DEBUG:nibe.nibe.connection.nibegw:Received 5c00206900499c from ('192.168.4.210', 10000)
DEBUG:nibe.nibe.connection.nibegw:Container: 
    address = (enum) MODBUS40 32
    cmd = (enum) MODBUS_READ_REQ 105
    data = b'' (total 0)
DEBUG:nibe.nibe.connection.nibegw:Received 5c00206c01014c0d from ('192.168.4.210', 10000)
DEBUG:nibe.nibe.connection.nibegw:Container: 
    address = (enum) MODBUS40 32
    cmd = (enum) MODBUS_WRITE_RESP 108
    data = Container: 
        result = True
INFO:nibe.nibe.connection.nibegw:Write succeeded for fan-mode-47260
```

yozik04, 2025-10-03:
Even if we will reopen, I have no idea what to do with it. Maybe in some states it is not allowed to write fan mode...

elupus, 2025-10-03:
We dont really know what the write return value implies to be honest. It could indicate accepted, but not applied yet.

I did see that for fault reset.. it will always deny same value writes.

plopp, 2025-10-03:
> It could actually be a value changed indication come to think of it.
Yes! That seems to be the case. Will test more, but so far your hypothesis holds.

yozik04, 2025-12-11:
I am thinking what would be returned if you try to write to readonly register, or try to write an invalid value. If same error, I would keep it at error level...