yozik04/nibe #183: Nibe s1256 .CSV upload in order to add entity for Hot water start (BT5)

KptnHaddock, 2024-10-16:
Posting the exported .CSV file from Nibe S1256 with reference to post [https://github.com/home-assistant/core/issues/128514](url)

Expored file: 
[modbus_addresses_all_20240901-06569924138015-2_en.csv](https://github.com/user-attachments/files/17396567/modbus_addresses_all_20240901-06569924138015-2_en.csv)

scruysberghs, 2024-12-19:
have you been able to connect over modbys tcp from home assistant using modbus tcp? I have a s1225 but can't get it to connect using (tcp://ip_adres_of_heat_pump:502 ) and modbus id 0

KptnHaddock, 2024-12-21:
@scruysberghs not sure I can help but what integration are you trying to use, is it [MQTT](https://www.home-assistant.io/integrations/mqtt), or [Nibe heatpump](https://www.home-assistant.io/integrations/mqtt)?  I successfully added the Nibe Heatpump integration for my s1256 and got a list of 1444 entities of which many can be enabled in HA. I got the integration to work via the Settings -> Integrations -> Add integration process. No editing required for that in the configuration.yaml file. 
Noticed also there is a later version on the [Modbus documentation](https://professional.nibe.eu/document/Technical%20information%20(TIF)/M12676EN.pdf) from Nibe that explains how to activate the mobdus connection from the heat pump.

tomasenskede, 2025-01-20:
This might be slightly off-topic, but how do you export all entities from an integration? Will it include entities that are not subscribed as well?

> Exported file: [modbus_addresses_all_20240901-06569924138015-2_en.csv](https://github.com/user-attachments/files/17396567/modbus_addresses_all_20240901-06569924138015-2_en.csv)

Regarding the missing GT5, I am also waiting for this sensor to appear in HA.

closed 2025-02-22 by yozik04