yozik04/nibe #189: Registers for SMO S40 (v3.3.8) + SPLIT HBS (v4.4) + RMU S40 (v2.0)

teixemf, 2024-11-18:
I recently installed an SMO S40 (software version 3.3.8) with a SPLIT HBS heat pump (software version 4.4) and a Room Monitoring Unit RMU S40 (software version 2.0).

I got the following Modbus registers from the USB interface: 
[modbus_addresses_all_20241117-1.csv](https://github.com/user-attachments/files/17794113/modbus_addresses_all_20241117-1.csv)

I started to work on merging the registers I got with the ones in [_/nibe/data/smos40.csv_](https://github.com/yozik04/nibe/blob/master/nibe/data/smos40.csv) but I realized some of my registers have different addresses than the ones on the repo.

How should this be handled?
Should I submit a PR anyway with these different values?

Here are all the registers I found with different addresses:

```
id:4045	MODBUS_HOLDING_REGISTER	5245	1	°C	1	1	40	8
id:4046	MODBUS_HOLDING_REGISTER	5247	1	°C	1	0	40	4
id:4049	MODBUS_HOLDING_REGISTER	5249	1		4	0	1	0
id:4050	MODBUS_HOLDING_REGISTER	5250	1		4	0	1	0
id:4051	MODBUS_HOLDING_REGISTER	5251	1		4	0	1	0
id:4591	MODBUS_HOLDING_REGISTER	5256	1		1	5	80	30
id:4659	MODBUS_HOLDING_REGISTER	5266	1		4	0	1	0
id:4660	MODBUS_HOLDING_REGISTER	5267	1		4	0	0	0
Shunted brine, accessory	MODBUS_HOLDING_REGISTER	5268	1		4	0	1	0
Shunted brine max. temp.	MODBUS_HOLDING_REGISTER	5271	1	°C	4	0	30	20
id:4703	MODBUS_HOLDING_REGISTER	5272	1		4	0	1	0
id:4704	MODBUS_HOLDING_REGISTER	5273	1		4	0	1	0
id:4705	MODBUS_HOLDING_REGISTER	5274	1		4	0	1	0
id:4706	MODBUS_HOLDING_REGISTER	5275	1		4	0	1	0
id:4707	MODBUS_HOLDING_REGISTER	5276	1		4	0	1	0
id:4708	MODBUS_HOLDING_REGISTER	5277	1		4	0	1	0
id:4709	MODBUS_HOLDING_REGISTER	5278	1		4	0	1	0
id:4710	MODBUS_HOLDING_REGISTER	5279	1		4	0	1	0
Input AUX6	MODBUS_HOLDING_REGISTER	767	1		4	0	67	0
id:4969	MODBUS_HOLDING_REGISTER	5295	1		4	0	1	0
id:4970	MODBUS_HOLDING_REGISTER	5296	1		4	0	1	0
id:4971	MODBUS_HOLDING_REGISTER	5297	1	Hz	4	20	117	20
id:4972	MODBUS_HOLDING_REGISTER	5298	1	Hz	4	20	117	20
id:4973	MODBUS_HOLDING_REGISTER	5299	1	Hz	4	23	120	23
id:4974	MODBUS_HOLDING_REGISTER	5300	1	Hz	4	23	120	23
Current power	MODBUS_INPUT_REGISTER	2727	100	kW	6	0	0	0
id:7048	MODBUS_HOLDING_REGISTER	5227	1		4	0	1	0
id:7138	MODBUS_HOLDING_REGISTER	5441	1	%	4	0	100	70
id:7139	MODBUS_HOLDING_REGISTER	5442	1		4	0	1	0
id:7140	MODBUS_HOLDING_REGISTER	5443	1		4	0	1	0
id:7141	MODBUS_HOLDING_REGISTER	5444	1	%	4	0	100	70
id:7142	MODBUS_HOLDING_REGISTER	5445	1		4	0	1	0
id:7143	MODBUS_HOLDING_REGISTER	5446	1		4	0	1	0
id:7144	MODBUS_HOLDING_REGISTER	5447	1	%	4	0	100	70
id:7145	MODBUS_HOLDING_REGISTER	5448	1		4	0	1	0
id:7146	MODBUS_HOLDING_REGISTER	5449	1		4	0	1	0
id:7147	MODBUS_HOLDING_REGISTER	5450	1	%	4	0	100	70
id:7148	MODBUS_HOLDING_REGISTER	5451	1		4	0	1	0
id:7149	MODBUS_HOLDING_REGISTER	5452	1		4	0	1	0
id:7150	MODBUS_HOLDING_REGISTER	5453	1	%	4	0	100	70
id:7176	MODBUS_HOLDING_REGISTER	5455	1		4	0	1	0
Minimum permitted speed (EB101 GP12)	MODBUS_HOLDING_REGISTER	5352	1	%	4	1	50	1
id:8982	MODBUS_HOLDING_REGISTER	5390	1		4	0	0	0
Audio signal on alarm	MODBUS_HOLDING_REGISTER	5326	1		4	0	1	1
Sound when pressing button	MODBUS_HOLDING_REGISTER	5327	1		4	0	1	1
BT12 offset, heat pump 1	MODBUS_HOLDING_REGISTER	5328	10	°C	1	-50	50	0
Start time	MODBUS_HOLDING_REGISTER	5393	1		-	-	-	-
Start time	MODBUS_HOLDING_REGISTER	5397	1		-	-	-	-
Stop time	MODBUS_HOLDING_REGISTER	5401	1		-	-	-	-
Disable emergency restart GP1	MODBUS_HOLDING_REGISTER	5392	1		4	0	1	0
Compressor, total time energy storage, main unit (EP14)	MODBUS_INPUT_REGISTER	2335	1	h	6	0	2147483647	0
Compressor, total time energy storage, main unit (EP15)	MODBUS_INPUT_REGISTER	2337	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 1 (EP14)	MODBUS_INPUT_REGISTER	2339	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 1 (EP15)	MODBUS_INPUT_REGISTER	2341	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 2 (EP14)	MODBUS_INPUT_REGISTER	2343	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 2 (EP15)	MODBUS_INPUT_REGISTER	2345	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 3 (EP14)	MODBUS_INPUT_REGISTER	2347	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 3 (EP15)	MODBUS_INPUT_REGISTER	2349	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 4 (EP14)	MODBUS_INPUT_REGISTER	2351	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 4 (EP15)	MODBUS_INPUT_REGISTER	2353	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 5 (EP14)	MODBUS_INPUT_REGISTER	2355	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 5 (EP15)	MODBUS_INPUT_REGISTER	2357	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 6 (EP14)	MODBUS_INPUT_REGISTER	2359	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 6 (EP15)	MODBUS_INPUT_REGISTER	2361	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 7 (EP14)	MODBUS_INPUT_REGISTER	2363	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 7 (EP15)	MODBUS_INPUT_REGISTER	2365	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 8 (EP14)	MODBUS_INPUT_REGISTER	2367	1	h	6	0	2147483647	0
Compressor, total time energy storage, heat pump 8 (EP15)	MODBUS_INPUT_REGISTER	2369	1	h	6	0	2147483647	0

```

yozik04, 2024-11-18:
We ignore all registers starting with "id: ".
For all the rest there will be id in the end of register name. So they should not conflict.

teixemf, 2024-11-30:
I have an RMU S40 connected to my system and I found out the Holding Register for the temperature set Point (0x03 2505) has its name starting with "id:" (id:12801).

Despite having its name starting with "id:" can I attempt a PR with a suggested a name for it since its purpose is known ?
(i.e. Temperature Set Point: RMU S40 1)

elupus, 2024-12-16:
Can you set values on that address? Would be cool to get virtual rmu over modbus.

elupus, 2024-12-16:
Btw. How did you find that address?

Ps. You can add stuff to extensions for this. This seems likely to exist for all S series pumps.

teixemf, 2024-12-17:
> Can you set values on that address? Would be cool to get virtual rmu over modbus.

Yes, I can set values on the Holding Register (0x03 2505) and they show up on my RMU S40 device screen.

> Btw. How did you find that address?

I started from downloading all registers from my SMO S40 via USB.
Then I compared my registers with the ones from issue #187 
The issue was open in early November and my SMO S40 is from late November. The registers could not be that different, so I thought. Although, I realized some registers on my file were missing on the file from issue #187 and vice-versa.
Some registers missing from the other file were related to my RMU S40 device.
I could tell this because my file add this perfectly identified entries:

```
Temperature: RMU S40 1 | MODBUS_INPUT_REGISTER | 2800 | 10 | °C | 3 | 0 | 0 | 0
Humidity: RMU S40 1 | MODBUS_INPUT_REGISTER | 3130 | 10 | %RH | 3 | 0 | 0 | 0
```

Despite these entries, I could not find any Holding Register related to the Temperature Set Point but I could identify a new Holding Register named "_id:12801_" with "_ºC_" Unit. 
I read it's value using a ModBus client and the value was the same I had set on my RMU S40 device.
I them changed the Temperature Set Point on my RMU S40 device to a different value and I got the same reading with the ModBus client. 
Further, I tried to write a value to that register, like I already said above, the value i wrote showed up on my RMU S40 device screen.

I then thought, maybe there are registers that only appear on the downloaded file.
I got to my SMO S40 settings menu and I activated a second RMU S40 unit.
Despite not having a second device, I could activate it on the SMO S40 settings menu.
The system thrown an alarm but I ignored it and I downloaded another registers file using the USB port.
When I opened the file I saw these new registers:

```
Temperature: RMU S40 2 | MODBUS_INPUT_REGISTER | 2801 | 10 | °C | 3 | 0 | 0 | 0
Humidity: RMU S40 2 | MODBUS_INPUT_REGISTER | 3131 | 10 | %RH | 3 | 0 | 0 | 0
```

The SMO S40 allows to add up to 8 RMU S40 devices.
I strongly believe it might be possible to use them as Virtual RMU. I somehow made the error go away and keep the second RMU on the SMO S40 screen, allowing me to adjust the Set Point.

I need to experiment with this a bit more bit this system is very new to me. 
I believe I need some guidance to validate what I am trying here.
And some advice not to brake the pump now since it's almost winter time.

> Ps. You can add stuff to extensions for this. This seems likely to exist for all S series pumps.

You mean, I should not had a a custom name to "_id:12801_" on the CSV file but on the extensions file ?

elupus, 2024-12-17:
Yup. Rmu are available on all s series. So extension file would make sense.

yozik04, 2025-02-22:
@teixemf Can you make a CSV export for me on FW 3.5.5?

closed 2025-02-22 by yozik04

yozik04, 2025-02-22:
No need. Found in a PR.