yozik04/nibe #203: new and changed register for S735-7

fritzschwalbe, 2025-01-13:
Hi Jevgeni
I'm using Homassistant to fetch the data and missing some registers of my Heatpump. I compared my register export with your existing file. On the one hand, there are new registers, including those that are no longer available, and on the other hand, there are changes to the current registers. The not anymore available registers doesn't disturb me and I will deactivate them in my Homassistant installation. The changed registers have new titles but the attributes are same. I'm not sure how to handle those. 
But could you at least add the new registers?

My environment:
**Home Assistant**
Core                             2025.1.2
Supervisor                  2024.12.3
Operating System     14.1
Frontend                     20250109.0

**Nibe S735-7 E EM 3x400V**
Software                      3.4.11
Version                         Pic61
Version                         805131
Version                         (MCB)99
Version                         (ZAB)99
Version                         (SFT)3
Supply air module      (SAM)27
Extra HW with solar   (DEW S43)

I uploaded 4 files:
[S735-7_E_EM-3.4.11.csv](https://github.com/user-attachments/files/18399786/S735-7_E_EM-3.4.11.csv)
My register export of the heatpump
[s735-COMPARISION.csv](https://github.com/user-attachments/files/18399787/s735-COMPARISION.csv)
Comaprision result between the current and my new export
[s735-NEW_REGISTER.csv](https://github.com/user-attachments/files/18399788/s735-NEW_REGISTER.csv)
updated csv file where changed registers kept as before and new registers added
[s735-CHANGED_and_NEW_REGISTER.csv](https://github.com/user-attachments/files/18399789/s735-CHANGED_and_NEW_REGISTER.csv)
updated csv file where changed registers got the new title new registers added

Many thanks for your support.
Best regards Peter

yozik04, 2025-01-13:
Great, now I will be able to act on it. Thank you!

yozik04, 2025-01-14:
It is sufficient just to give a register import as I will do diff in VS Code.

fritzschwalbe, 2025-02-10:
Hello @yozik04 
Hello @elupus 
I have seen that the change of the register is already taken over in the Master Branch, how is the new register template taken over in Homeassistant? 
In Homeassistant I have all previous updates and am today on HA Core 2025.2.1
Sorry to bother you again.

closed 2025-02-22 by yozik04