yozik04/nibe #209: Adding missing modbus registers for blockFreq on S2125 / VVM S320

martijnrusschen, 2025-01-23:
I'm trying control the blockFreq settings that are available on my Nibe heatpump (S2125-12 & VVM S320). According to the modbus registry, there should be modbus registers available for this, so I'm wondering if these could be added to the integration. Ideally, I'd be able to turn on/off the setting, and control the 2 settings that control which frequencies are blocked.

Updated - voor S2125-12(Version: 3.1.1) met SMOS40(Version: 3.5.5)
BlockFreq. 1 On/Off
Title in modbus export: id:4970
Register: 5296

BlockFreq.1 from
Title in modbus export: id:4972
Register: 5298

BlockFreq.1 to
Title in modbus export: id:4974
Register: 5300

martijnrusschen, 2025-01-23:
I was also looking into how I could export the registers myself, but wasn't sure how this works on the S series. I didn't see the option in the menu at least.

elupus, 2025-01-23:
The exports only list features enabled in the pump. Which is why we need exports from users who as the feature enabled.

martijnrusschen, 2025-01-23:
Gotcha, is there a guide available on how to export the registers?

elupus, 2025-01-23:
Somewhere in the pump menus where you enable modbus i think. Used to have a link to a document, but cant find it anymore.

Doorshifter, 2025-01-23:
https://assetstore.nibe.se/hcms/v2.3/entity/document/878103/storage/ODc4MTAzLzAvbWFzdGVy 
On page 4, the ‘tips’ is what you are looking for.

martijnrusschen, 2025-01-30:
Any chance we can run a release soon?