yozik04/nibe #215: Add Energy Production coils to SMO40

tanelvakker, 2025-02-13:
VVM320 has these registers:

"Heat Meter - HW Cpr EP14";"Accumulated energy production as calculated by the heat meter";44306;"kWh";u32;10;0;0;0;R;
"Heat Meter - Heat Cpr EP14";"Accumulated energy production as calculated by the heat meter";44308;"kWh";u32;10;0;0;0;R;

They are missing on SMO40 and the Nibe provided tool does not output them for some reason.
Can these be added to SMO40 as well. They were tested to exist on SMO40 with nibepi.

For context, I will use them for COP measurement.

elupus, 2025-02-13:
They only provide valid data if you have a flow meter installed (which som countries have by default).

Yes those could be added. The extension file:  https://github.com/yozik04/nibe/blob/master/nibe/data/extensions.json would need to be modified.

closed 2025-02-22 by yozik04

gaaf, 2025-03-04:
I don't think these are any use as AFAIK the SMO40 has no option to connect flow- or heatmeters.

commit e6efbbdc (2025-02-22): Closes #215