yozik04/nibe #218: Add human readable alarm text

tizianodeg, 2025-03-09:
## Feature Request

**Heatpump model**:
F730 ...

**Firmware version**:

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

I would like to add human readable alarm messages to home assistant. 
[image]

**Describe alternatives you've considered**
From this pdf the meaning of error messages can be taken:
https://installer.nibe.eu/download/18.17fb7f4e185eab8ed4027e7/1676466235274/Alarm%20list%20Emmy%201747-1.pdf

I think the best would be to keep the current coil alarm_45001 as integer as people might use it already.
Not sure about the best approach, so would love to have a peer with a suggestion.

My first approach would be:
a) just add translations to the error codes of that coil in home assistant nibe component
b) add a "virtual coil" alarm_45001_text with the mapped text. 

I would prefer to go with option a) 
see my first try:
https://github.com/home-assistant/core/commit/a2ffcb2d51e6d16791dd27961808814f5c6f40e3

[image]

hass-tiziano, 2025-05-14:
@elupus @yozik04 hey guys, sorry for the ping, I know everyone has a real life beside of such fun projects 😅

I would love to have a quick feedback, the change is pretty small you can see it in my change commit, will cost you less than a minute and maybe 5minute thinking of a better solution.

If not I will try to directly create a PR and reference you 😬

Looking forward to see readable errors on my NIBE dashboard 😉

Have a nice week!
Tiziano

elupus, 2025-05-14:
Easier to do if you open a pull request :).

yozik04, 2025-05-14:
Looked into the commit. This will not fly. We have many heat pump models, all can have different error codes with different error descriptions. This requires more design.

hass-tiziano, 2025-05-14:
I see... semantics... but I did not find other error codes and could not test them neither.
This list worked for my F730.

So your suggestion would be to add a virtual coil to the heat pumps that matches the error codes. I think at least F750 and F730. This would then need to be done in this repo and not in home assistant.

hass-tiziano, 2025-05-14:
@yozik04 you are totally right...

https://www.nibe.eu/ch/de/support/fehlermeldungen

Same error codes have different meaning for F ands S series 🙈